360Controller/360Controller #958: Can't install

WaddlesM, 2019-10-26:
My Issue is that after the restart i went to privicy settings but it wasnt there or it was just i couldnt find it. Can you say where and what the file name is please? Thanks!

FranticRain, 2019-10-26:
If you open your system preferences do you see a new preference pane for the driver? If you do, press the "enable driver" button, as it will begin a debug process that will help you figure out what's gone wrong. Otherwise, I'm going to need you to fill out the issue template.

WaddlesM, 2019-10-26:
Thanks! But Theres no new preference pane for the driver they all look the same can you send me a screenshot of what it looks like? (i dont know how to fill out issue template

FranticRain, 2019-10-26:
If you don't see one that explicitly says something like "360 Controller" then you didn't install it. Try running the installer again.

WaddlesM, 2019-10-26:
k ill do

WaddlesM, 2019-10-26:
Nevermind it was already installed from me installing it i just didnt check (CLOSED)

WaddlesM, 2019-10-26:
Wait finds wired controller but cant see buuton presses and cant find wireless? can you give me a updated version that fixes that 
Controller:Xbox one S/X White
Os:Mac Beta 1.15
Chargher that works:Anker Teared
Please send the updated one (if you can) At waddlesbergeron@gmail.com

FranticRain, 2019-10-26:
I can't just give you a version that "fixes" your problem and email it to you if I don't know what your problem is. There is no macOS 1.15, and if you mean 10.15, it's not in beta anymore, and if you're using a beta version of macOS, I'm definitely not going to fix it for you. I only fix the driver for stable versions of macOS. I don't know what the product and vendor IDs of your controller are. I don't know what you mean by the charger that worked. It should just be a micro usb cable. Here's the full template, fill it out in full or I am just going to lock this issue.

```
### Type of Controller
Replace this text with: Original Xbox, Xbox 360, Xbox One, All or N/A

### OS Version
Replace this text with the version of macOS that you are running

### Driver Version
Replate this text with the version or versions of the driver that you are using or have tested

### Connection Method
Replace this text with: Wired, Wireless Adapter, Bluetooth, N/A or more specific method if applicable

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID:
<br/>Vendor ID:

### Response of Controller
Replace this text with information on how the physical controller reacts. Does the guide button light up? Does it vibrate? Etc.

### Response in Preference Pane
Replace this text with information about the controller's reaction in the preference pane. Does the controller appear in the preference pane? Do all of the buttons, joysticks, and triggers act as expected?

### Detailed Explaination of Issue
Replace this text with a detailed explaination of your issue. Simply saying "it doesn't work" is a quick way to ensure that your issue is never fixed. Without detailed information on your specific issue, we cannot fix your issue. Including as much information as possible in this section increases the likelyhood of getting your issue quickly and effectively resolved.
```

WaddlesM, 2019-10-28:
### Type of Controller
Xbox One

### OS Version
10.15 (Not Stable)

### Driver Version
N/A

### Connection Method
Wired/Wireless

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
N/A

### Response of Controller
Goes Crazy Presses Every button it can in some games

### Response in Preference Pane
Nothing says its Wired but nothing works

### Detailed Explaination of Issue
Wired Conntroller Says Is Connected But Not and presses every button at 1 time 

You’re receiving notifications because you modified the open/close state.
2 participants
@WaddlesM
@FranticRain

WaddlesM, 2019-10-28:
If You want me to close this ill close it after school hours (7:30AM-4:00PM EST)

FranticRain, 2019-10-28:
I'm closing this issue because first off, you are running a beta version of macOS and I don't support those. Second, you didn't provide your device name and info, which is *absolutely crucial* to solving this problem.

closed 2019-10-28 by FranticRain