360Controller/360Controller #959: Driver hangs when enabling

akiosaito, 2019-10-27:
### Type of Controller
Xbox One

### OS Version
Mac OS X 10.14.6 (Build 18G95)

### Driver Version
360Controller 1.0.0-alpha.4

### Connection Method
Wired

### Device Name and Info
>Device Name: Controller
<br/>Product ID: 0x02ea
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller
N/A

### Response in Preference Pane
N/A

### Detailed Explaination of Issue
Driver (System Preference) always hangs when trying to "Enable Driver". I've attached a system report for any cases.

Thanks in advance!
[360cont_hang_report.txt](https://github.com/360Controller/360Controller/files/3775401/360cont_hang_report.txt)

FranticRain, 2019-10-27:
Okay so long story short, when you’re pressing the “enable driver” button, it doesn’t actually enable the driver. Rather, it does a debug process to try and diagnose issues with your driver installation. Clearly there’s been some sort of issue, because the driver should be loaded automatically when you plug in a controller, and it shouldn’t crash when you run the diagnostic. So let’s run it by hand and see if we can figure out what the real problem is. Run this command in your terminal and let me know what the output is. It will prompt you for your password, but this is just because it’s a macOS tool that is trying to manually load the driver.

```
sudo kextutil /Library/Extensions/360Controller.kext
```

akiosaito, 2019-10-27:
Thanks for the prompt reply!

`$ sudo kextutil /Library/Extensions/360Controller.kext
Password:
Error copying bundle during staging: -1
Unable to stage kext (/System/Library/Extensions/roller.kext) to secure location.
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

(kernel) kxld[com.mice.driver.Xbox360Controller]: The super class vtable '__ZTV9IOService' for vtable '__ZTV17Xbox360Peripheral' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The super class vtable '__ZTV11IOHIDDevice' for vtable '__ZTV20ChatPadKeyboardClass' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The super class vtable '__ZTV11IOHIDDevice' for vtable '__ZTV22Xbox360ControllerClass' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV22Xbox360Pretend360Class was not patched because its parent, the __ZTV22Xbox360ControllerClass, was not patchable.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV22XboxOneControllerClass was not patched because its parent, the __ZTV22Xbox360ControllerClass, was not patchable.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV22XboxOnePretend360Class was not patched because its parent, the __ZTV22XboxOneControllerClass, was not patchable.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV27XboxOriginalControllerClass was not patched because its parent, the __ZTV22Xbox360ControllerClass, was not patchable.
(kernel) Can't load kext com.mice.driver.Xbox360Controller - link failed.
(kernel) Failed to load executable for kext com.mice.driver.Xbox360Controller.
(kernel) Kext com.mice.driver.Xbox360Controller failed to load (0xdc008016).
(kernel) Failed to load kext com.mice.driver.Xbox360Controller (error 0xdc008016).
Failed to load /Library/StagedExtensions/Library/Extensions/360Controller.kext - (libkern/kext) link error.
Check library declarations for your kext with kextlibs(8).
`

FranticRain, 2019-10-27:
This is a known issue, thankfully. I’m working on a fix and hope to have it out soon.

kud, 2019-11-03:
Yeah I've got this too:

[image]

FranticRain, 2019-11-03:
@kud Thats the same issue. Will be covered in the fix.

FranticRain, 2019-11-04:
This should be resolved in alpha 5. Let me know if it fixes your issue.

kud, 2019-11-04:
I'll do, cheers!

akiosaito, 2019-11-05:
Confirmed the fix.
Thanks  a lot!

closed 2019-11-05 by akiosaito