360Controller/360Controller #960: PowerA - Xbox One controller not recognised

tumidrain880899, 2019-11-01:
### Type of Controller
PowerA Xbox One Controller
### OS Version
10.14 Mojave

### Driver Version
360Controller 1.0.0-alpha.4
### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: XB1 Classic Controller
<br/>Product ID: 0x581a
<br/>Vendor ID: 0x24c6

### Response of Controller
Does not light up

### Response in Preference Pane
Menu listed a device called XB1 Classic Controller

### Detailed Explaination of Issue
It doesn't work

FranticRain, 2019-11-04:
Can you check the "enable driver" checkbox in the advanced settings and follow any debug steps that it provides? If it gives you an error, let me know.

tumidrain880899, 2019-11-04:
It can be used now!!! Thank you so much

closed 2019-11-04 by FranticRain