360Controller/360Controller #961: Xbox One Controller Not Found

Chris-Anime, 2019-11-04:
### Type of Controller
Xbox One

### OS Version
macOS Mojave
10.14.6

### Driver Version
1.0.0-Alpha-4

### Connection Method
Wired

### Device Name and Info
>Device Name: N/A
<br/>Product ID: 0x02ea
<br/>Vendor ID: 0x045e

### Response of Controller
Controller Vibrates, but does not light up.

### Response in Preference Pane
Does not act at all

### Detailed Explaination of Issue
I've tried connecting using different USB cables to see if it's the charging cable vs data cable, and i'm certain i'm using a data cable. I think I allowed the driver in preferences, but i'm not 100 percent sure. I'm not sure how to solve the issue, and I've been looking through other posts, but not many line up with my macOS.

FranticRain, 2019-11-04:
Can you go into the advanced settings of the driver and check the "enable driver" checkbox? It will provide you will some debug information that should help you resolve your issues. Let me know if you get any errors from it.

Chris-Anime, 2019-11-04:
Ok, so I tried to click enable driver and got this message,
"The driver has failed to load due to an unknown error, code: -603946986. Please report this error to the driver maintainer."

FranticRain, 2019-11-04:
This is a known issue and will be fixed in the next version. For now, alpha 3 should work for you.

Chris-Anime, 2019-11-04:
Ok, thanks for the help

FranticRain, 2019-11-04:
This should be resolved in alpha 5. Let me know if it works for you.

Tzhies, 2019-11-04:
Hi FranticRain,
I just downloaded the alpha 5 and still got the described problem with the driver saying "Cannot load driver" and the following error message.

tarokker, 2019-11-06:
Hi, I don't know if it's ok for me to join the topic or if you prefer to open a new issue:

it seems like I have the same problem with my Xbox one Controller: I attach the cable, the controller bibrates, but does not light up.

I started on Mojave with alpha4 and the infamous error ""The driver has failed to load due to an unknown error, code: -60394698".

Then I uninstalled the driver, installed alpha5 and the error went away, but the controller still didn't light up.

This morning I uninstalled the driver, installed Catalina e reinstalled the alpha5 driver, but still it doesn't light up :(

Please let me know if I could help you with some debug or logs..

Thanks in advance

FranticRain, 2019-11-06:
@tarokker I'd appreciate it if you could make a new issue just so I can get all of the template data from you. It helps a lot in solving these sorts of issues. Also, if you could, go to the "advanced" tab of the preference pane and check the "enable driver" button. I'm sure you know that it'll give you some debug output to try if something is wrong. I'd love to know if it's giving you any errors or not.

Joshuasony, 2019-12-25:
I'll join this discussion since I had the same issue but now it works with Alpha 3. 
First I installed the newest version Alpha 5. Wouldn't work. Then Alpha 4, also wouldn't work. 
First my issue was that the controllers light would only light quickly after plugin in the controller. Then the light would disappear and the controller wasn't listed in the settings. With Alpha 4 it was constantly lighting up (Player 1), but still not listed. With Alpha 3 Player 1 is constantly lighting up and the controller was detected. 
I'm using a wired glow controller.

FranticRain, 2019-12-25:
@Joshuasony Mind giving this build a shot? 

[360ControllerInstall.dmg.zip](https://github.com/360Controller/360Controller/files/3993107/360ControllerInstall.dmg.zip)

Joshuasony, 2019-12-25:
Hi @FranticRain, this seems to work as well. Did you find the cause of the issue?

FranticRain, 2019-12-25:
@Joshuasony Yup. It was the way I was building the driver. Catalina messed up my whole build system so I had some bad builds that I put up.

Going to go ahead and close this one, since the fixed build is out there now.

closed 2019-12-25 by FranticRain