360Controller/360Controller #963: Installation fails on Catalina

PokeCaptain, 2019-11-04:
### Type of Controller
N/A
### OS Version
10.15.1

### Driver Version
1.0.0α5
### Connection Method
N/A

### Device Name and Info
N/A

### Response of Controller
N/A

### Response in Preference Pane
Pref pane does not show up after installation attempt

### Detailed Explaination of Issue
Ran uninstall script to make sure I had no old driver code left, then ran the new pkg. Install log is as follows:

`2019-11-04 12:16:51-05 macbook-pro-3 Installer[27584]: Opened from: /Volumes/360ControllerInstall/Install360Controller.pkg
2019-11-04 12:16:51-05 macbook-pro-3 Installer[27584]: Package Authoring Error: <background_scaling> has an unsupported MIME type: X-NSObject/NSNumber
2019-11-04 12:16:51-05 macbook-pro-3 Installer[27584]: Package Authoring Error: <background_alignment> has an unsupported MIME type: X-NSObject/NSNumber
2019-11-04 12:16:51-05 macbook-pro-3 Installer[27584]: Package Authoring Error: <layout-direction> has an unsupported MIME type: X-NSObject/NSNumber
2019-11-04 12:16:51-05 macbook-pro-3 Installer[27584]: Failed to load specified background image
2019-11-04 12:16:51-05 macbook-pro-3 Installer[27584]: Product archive /Volumes/360ControllerInstall/Install360Controller.pkg trustLevel=350
2019-11-04 12:16:51-05 macbook-pro-3 Installer[27584]: External component packages (1) trustLevel=350
2019-11-04 12:16:59-05 macbook-pro-3 Installer[27584]: Administrator authorization granted.
2019-11-04 12:17:01-05 macbook-pro-3 Installer[27584]: ================================================================================
2019-11-04 12:17:01-05 macbook-pro-3 Installer[27584]: User picked Standard Install
2019-11-04 12:17:01-05 macbook-pro-3 Installer[27584]: Choices selected for installation:
2019-11-04 12:17:01-05 macbook-pro-3 Installer[27584]: 	Install: "Install360Controller"
2019-11-04 12:17:01-05 macbook-pro-3 Installer[27584]: 	Install: "Install360Controller"
2019-11-04 12:17:01-05 macbook-pro-3 Installer[27584]: 		Install360Controller.pkg#Install360Controller.pkg : com.mice.pkg.Xbox360Controller : 1.0.0-alpha.5
2019-11-04 12:17:01-05 macbook-pro-3 Installer[27584]: ================================================================================
2019-11-04 12:17:01-05 macbook-pro-3 Installer[27584]: It took 0.00 seconds to summarize the package selections.
2019-11-04 12:17:01-05 macbook-pro-3 Installer[27584]: -[IFDInstallController(Private) _buildInstallPlanReturningError:]: location = file://localhost
2019-11-04 12:17:01-05 macbook-pro-3 Installer[27584]: -[IFDInstallController(Private) _buildInstallPlanReturningError:]: file://localhost/Volumes/360ControllerInstall/Install360Controller.pkg#Install360Controller.pkg
2019-11-04 12:17:01-05 macbook-pro-3 Installer[27584]: Set authorization level to root for session
2019-11-04 12:17:01-05 macbook-pro-3 Installer[27584]: Will use PK session
2019-11-04 12:17:01-05 macbook-pro-3 Installer[27584]: Using authorization level of root for IFPKInstallElement
2019-11-04 12:17:01-05 macbook-pro-3 Installer[27584]: Starting installation:
2019-11-04 12:17:01-05 macbook-pro-3 Installer[27584]: Configuring volume "Macintosh HD"
2019-11-04 12:17:01-05 macbook-pro-3 Installer[27584]: Preparing disk for local booted install.
2019-11-04 12:17:01-05 macbook-pro-3 Installer[27584]: Free space on "Macintosh HD": 30.61 GB (30606180352 bytes).
2019-11-04 12:17:01-05 macbook-pro-3 Installer[27584]: Create temporary directory "/var/folders/8y/_mq__w4j4jd4xrjgq1wqsnjr0000gn/T//Install.27584kkPl3b"
2019-11-04 12:17:01-05 macbook-pro-3 Installer[27584]: IFPKInstallElement (1 packages)
2019-11-04 12:17:01-05 macbook-pro-3 installd[595]: PackageKit: Adding client PKInstallDaemonClient pid=27584, uid=501 (/System/Library/CoreServices/Installer.app/Contents/MacOS/Installer)
2019-11-04 12:17:01-05 macbook-pro-3 Installer[27584]: PackageKit: Enqueuing install with framework-specified quality of service (utility)
2019-11-04 12:17:01-05 macbook-pro-3 installd[595]: PackageKit: ----- Begin install -----
2019-11-04 12:17:01-05 macbook-pro-3 installd[595]: PackageKit: request=PKInstallRequest <1 packages, destination=/>
2019-11-04 12:17:01-05 macbook-pro-3 installd[595]: PackageKit: packages=(
	    "PKLeopardPackage <id=com.mice.pkg.Xbox360Controller, version=1.0.0-alpha.5, url=file:///Volumes/360ControllerInstall/Install360Controller.pkg#Install360Controller.pkg>"
	)
2019-11-04 12:17:01-05 macbook-pro-3 installd[595]: PackageKit: Set reponsibility for install to 27584
2019-11-04 12:17:01-05 macbook-pro-3 installd[595]: PackageKit: Skipping stale sandbox at path /Library/InstallerSandboxes/.PKInstallSandboxManager/F5AC417B-1CA7-4FE7-BCFC-BFFB12FAAE6F.activeSandbox
2019-11-04 12:17:01-05 macbook-pro-3 installd[595]: PackageKit: Extracting file:///Volumes/360ControllerInstall/Install360Controller.pkg#Install360Controller.pkg (destination=/Library/InstallerSandboxes/.PKInstallSandboxManager/B89FA420-CA83-4FA0-93EB-AB966EBAF0EA.activeSandbox/Root, uid=0)
2019-11-04 12:17:01-05 macbook-pro-3 installd[595]: PackageKit: prevent user idle system sleep
2019-11-04 12:17:01-05 macbook-pro-3 installd[595]: PackageKit: suspending Spotlight indexing
2019-11-04 12:17:01-05 macbook-pro-3 installd[595]: PackageKit: suspending backupd
2019-11-04 12:17:01-05 macbook-pro-3 installd[595]: PackageKit: Executing script "./preinstall" in /private/tmp/PKInstallSandbox.Q1Y64B/Scripts/com.mice.pkg.Xbox360Controller.0KOsb3
2019-11-04 12:17:02-05 macbook-pro-3 installd[595]: ./preinstall: No receipt for 'com.mice.pkg.Xbox360Controller' found at '/'.
2019-11-04 12:17:02-05 macbook-pro-3 installd[595]: PackageKit: Using trashcan path /var/folders/zz/zyxvpxvq6csfxvn_n0000000000000/T/PKInstallSandboxTrash/B89FA420-CA83-4FA0-93EB-AB966EBAF0EA.sandboxTrash for sandbox /Library/InstallerSandboxes/.PKInstallSandboxManager/B89FA420-CA83-4FA0-93EB-AB966EBAF0EA.activeSandbox
2019-11-04 12:17:02-05 macbook-pro-3 installd[595]: PackageKit: Shoving /Library/InstallerSandboxes/.PKInstallSandboxManager/B89FA420-CA83-4FA0-93EB-AB966EBAF0EA.activeSandbox/Root (1 items) to /
2019-11-04 12:17:02-05 macbook-pro-3 shove[27593]: [source=dir, dst=file] failed unlink(/Library/Extensions/360Controller.kext): Operation not permitted
2019-11-04 12:17:02-05 macbook-pro-3 shove[27593]: dstPath = /Library/Extensions/360Controller.kext NSFileOwnerAccountID=0 NSFileHFSTypeCode=1936485995 NSFileSystemFileNumber=833817 NSFileExtensionHidden=0 NSFileSystemNumber=16777222 NSFileSize=38 NSFileGroupOwnerAccountID=0 NSFileHFSCreatorCode=1919443312 NSFileCreationDate=2017-05-16 02:11:40 +0000 NSFileOwnerAccountName=root NSFilePosixPermissions=493 NSFileType=NSFileTypeSymbolicLink NSFileGroupOwnerAccountName=wheel NSFileReferenceCount=1 NSFileModificationDate=2017-05-16 02:11:40 +0000 XATTR=com.apple.rootless
2019-11-04 12:17:02-05 macbook-pro-3 installd[595]: PackageKit: releasing Spotlight indexing
2019-11-04 12:17:02-05 macbook-pro-3 installd[595]: PackageKit: releasing backupd
[... 7 lines cut ...]
	    "/"
	), NSUnderlyingError=0x7ff472603a10 {Error Domain=NSPOSIXErrorDomain Code=1 "Operation not permitted"}} {
	    NSLocalizedDescription = "An unexpected error occurred while moving files to the final destination.";
	    NSUnderlyingError = "Error Domain=NSPOSIXErrorDomain Code=1 \"Operation not permitted\"";
	    arguments =     (
	        "-f",
	        "-s",
	        "/Library/InstallerSandboxes/.PKInstallSandboxManager/B89FA420-CA83-4FA0-93EB-AB966EBAF0EA.activeSandbox/Root",
	        "/"
	    );
	}
2019-11-04 12:17:02-05 macbook-pro-3 installd[595]: PackageKit: Running idle tasks
2019-11-04 12:17:02-05 macbook-pro-3 installd[595]: PackageKit: Removing client PKInstallDaemonClient pid=27584, uid=501 (/System/Library/CoreServices/Installer.app/Contents/MacOS/Installer)
2019-11-04 12:17:02-05 macbook-pro-3 Installer[27584]: install:didFailWithError:Error Domain=PKInstallErrorDomain Code=120 "An unexpected error occurred while moving files to the final destination." UserInfo={NSUnderlyingError=0x600001ead800 {Error Domain=NSPOSIXErrorDomain Code=1 "Operation not permitted"}, NSLocalizedDescription=An unexpected error occurred while moving files to the final destination., arguments=(
	    "-f",
	    "-s",
	    "/Library/InstallerSandboxes/.PKInstallSandboxManager/B89FA420-CA83-4FA0-93EB-AB966EBAF0EA.activeSandbox/Root",
	    "/"
	)}
2019-11-04 12:17:02-05 macbook-pro-3 installd[595]: PackageKit: Done with sandbox removals
[... cut ...] d an error that caused the installation to fail. Contact the software manufacturer for assistance.
2019-11-04 12:17:03-05 macbook-pro-3 Installer[27584]: IFDInstallController 109C5C0 state = 8
2019-11-04 12:17:03-05 macbook-pro-3 Installer[27584]: Displaying 'Install Failed' UI.
2019-11-04 12:17:03-05 macbook-pro-3 Installer[27584]: 'Install Failed' UI displayed message:'The Installer encountered an error that caused the installation to fail. Contact the software manufacturer for assistance.'.
`

FranticRain, 2019-11-04:
Did you encounter this same issue with alpha 4? They should be functionally identical for Catalina installs. You should be able to at least use the driver in alpha 4 while this gets solved.

PokeCaptain, 2019-11-04:
Same issue occurs with alpha 4.

FranticRain, 2019-11-04:
```
Operation not permitted 2019-11-04 12:17:02-05 macbook-pro-3 shove[27593]: dstPath = /Library/Extensions/360Controller.kext NSFileOwnerAccountID=0 NSFileHFSTypeCode=1936485995 NSFileSystemFileNumber=833817 NSFileExtensionHidden=0 NSFileSystemNumber=16777222 NSFileSize=38 NSFileGroupOwnerAccountID=0 NSFileHFSCreatorCode=1919443312 NSFileCreationDate=2017-05-16 02:11:40 +0000 NSFileOwnerAccountName=root NSFilePosixPermissions=493 NSFileType=NSFileTypeSymbolicLink NSFileGroupOwnerAccountName=wheel NSFileReferenceCount=1 NSFileModificationDate=2017-05-16 02:11:40 +0000 XATTR=com.apple.rootless
```
Seems like you don't have access to edit your `/Library` folder. Are you using an administrator account on your PC?

PokeCaptain, 2019-11-04:
Administrator account is being used. /Library seems to be owned by 'system' and I cannot change the owner or add writable users. Attempted to do so with both Finder and sudo chown. It might be related to the new read-only system volume in Catalina. [This](https://eclecticlight.co/2019/10/11/macos-catalina-boot-volume-layout-revised-for-10-15-release/l) post may be helpful. I haven't checked the install script myself to see if this is the case, but it's the only thing I could think of that would cause problems

FranticRain, 2019-11-04:
I don’t think you linked the right thing there. But I was under the impression that the read-only file system was limited to `/System` and had nothing to do with `/Library`.

PokeCaptain, 2019-11-04:
I think I edited the link right. Title should be "macOS Catalina Boot Volume Layout revised for 10.15 release". The read only fs seems to include a lot more than just `/System` if I'm reading that right