360Controller/360Controller #966: Xbox Elite v2 Controller Support

edolyne, 2019-11-05:
### Type of Controller
Xbox Elite v2

### OS Version
10.15.1

### Driver Version
1.0.0-alpha.5

### Connection Method
Bluetooth, via USB the controller is not recognized

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID: 0x0B05
<br/>Vendor ID: 0x045E

### Response of Controller
When connected the guide button is constantly lit.

### Response in Preference Pane
In the preference pane it says no devices found

### Detailed Explaination of Issue
Under advanced I've checked Enable Driver, however there is no controller recognized, the deadzones doesn't provide any feedback and on controller test no buttons identify themselves when pressed on the controller. So while the controller is paired and connected via bluetooth there is no input being registered on the computer.

FranticRain, 2019-11-12:
Here's a build with your controller added. Let me know how it works. I want to make sure that the new elite controller works as expected.

[360ControllerInstall.dmg.zip](https://github.com/360Controller/360Controller/files/3834377/360ControllerInstall.dmg.zip)

grp, 2019-11-13:
Hi, I didn't open this issue but I did try the build linked above. It doesn't work for me, although my controller is listed with a different product ID:

    Product ID:   0x0b00
    Vendor ID:    0x045e

It is also an Xbox Elite Wireless Controller 2, so maybe there are multiple product IDs for it.

grp, 2019-11-13:
I tried a local build with the product ID above added. It detects the controller, but no input works. I am on Catalina so that may be #968.

FranticRain, 2019-11-13:
It’s far more likely that there is a new byte packet than a duplicate of that issue. Do you have any other Xbox controllers you could test with the driver? Because if it is only the elite controller that doesn’t work, it’s the byte packet. If it’s for both, then it’s a duplicate to that issue.

mattmactimpson, 2019-11-13:
Just a quick question 
I don’t know if this is possible 
But is there a way to extract the controller support from Catalina and migrate it into this program ? 
I don’t know if it would only be possible on Mojave or is it possible at all 
Sorry I’m not the most code savvy person

mattmactimpson, 2019-11-13:
I have my xbox one x refresh controller that works on usb but not the elite series 2 if that helps on info

grp, 2019-11-13:
I don’t, unfortunately. Maybe someone else with this controller does (or isn’t on Catalina) and could figure that out.

mattmactimpson, 2019-11-13:
I am running Mojave and have 2 controllers the series 2 does not work the one x controller does work

FranticRain, 2019-11-13:
@mattmactimpson This is impossible and out of scope. Bluetooth controllers are supported natively through macOS and not within the purview of this driver. As for your wired issues, are you using the the driver version included in this thread? Can you confirm the vendor and product IDs of your controller? Instructions are included in the README of this project.

@grp My current guess is that it's a different packet format. I want to get some more information to make sure. If you could, and if you have a windows PC available, I need a USB packet capture. The process is listed here: http://vestigl.com/sniffing/index.html I just need the button press packets, so you don't need to worry about getting the connection logic. Just plug the controller in to a port, start the sniff, press some buttons, and then end the sniff.

mattmactimpson, 2019-11-13:
Bluetooth Xbox controllers are not supported on Mac OS Mojave
Yes I’m using current driver 
Vendor id 0x045e
Product id 0x0b00

FranticRain, 2019-11-13:
@mattmactimpson Okay. That's definitely not going to work because that ID isn't recognized by the driver. I'll get a special version out to you soon that *should* work with your controller.

FranticRain, 2019-11-14:
Here's a build that should include support for that `0x0b00` version of the Elite controller. Let me know how it works.

[360ControllerInstall.dmg.zip](https://github.com/360Controller/360Controller/files/3844619/360ControllerInstall.dmg.zip)

grp, 2019-11-14:
Probably expected, but the new build has the same result as my local build. I'll see if I can borrow a Windows PC to run the capture.

FranticRain, 2019-11-14:
@grp Yeah. I didn't anticipate that would work, but the kernel works in mysterious ways, so I wanted to check.

To anyone with this controller, if you can get me a packet sniff, that'd be amazing. It isn't going to be feasible to solve this without it. Dunno why Microsoft changed the packets again, but you know. Gotta make some more work for me, right?

b3ll, 2019-11-16:
hey @grp and @FranticRain I went and dumped the controller, I did two dumps to be sure (I think I screwed up the button presses on the first one)

[controller_dumps.zip](https://github.com/360Controller/360Controller/files/3853815/controller_dumps.zip)

FranticRain, 2019-11-16:
@b3ll Thanks for that. I should be able to get it working with this.

In case anyone was curious what this looks like:
It's a similar style of packet to the original elite controller, but now contains 5 extra bytes of data. None of the sniffs provided any insight into what this extra data was because it was all `00`. Similarly, the original elite controller contained a sort of "true" value in the secondary part of the packet, where it would record the real value from the controller and not the mapped one. For example, if you had remapped your A button to LB, then the first part of the packet would report LB and the second part would report A when you pressed the A button on your gamepad. However, in these sniffs, the second half of the packet is 0-ed out. Even the sticks. It would probably require some more investigation to figure out what this means for the actual contents of the packets, but for now I'm not particularly worried about it.

FranticRain, 2019-11-16:
In reviewing the code, these changes to the byte packet shouldn't be preventing the controller from properly passing data through the driver. I'm a little bit stumped. Weird question, but can you get the guide button to report a press at all?

sNohz, 2019-11-29:
Hi, same problem here, is there anything I can do to help ?

FranticRain, 2019-11-29:
@sNohz If you could answer the question above, using one of the drivers linked in this thread, that might help. Does pressing the guide button ever show up in the preference pane in macOS?

sNohz, 2019-11-29:
Oh ! I just reinstalled the build from 16 days ago and this time all the controller is working

FranticRain, 2019-11-29:
@sNohz Can you confirm what the product ID of your controller is? If you don’t know how to find this, there is a guide in the README.

To everyone else: I am going to assume that this problem is solved and close it in the coming days unless anyone can confirm that they are having issues with the build I posted in this thread.

sNohz, 2019-11-29:
I have 
`Product ID:   0x0b00`

newcontender, 2019-11-30:
Hello,

I tried the build from 16 days ago and My Elite 2 controller is detected now but none of the inputs are being registered in the controller test. Let me know if you require any further details.

FranticRain, 2019-11-30:
@newcontender What is the product ID of your controller? And have you checked every profile on your controller?

newcontender, 2019-11-30:
Hi,

I have tried all the profiles on the controller. Here is the Product and Vendor ID:

Product ID:	0x0b00
  Vendor ID:	0x045e

FranticRain, 2019-12-01:
Okay so now we have two identical controllers with two different results. Have you tried updating the firmware on your controller? Just to make sure, since this seems like the only difference now.

newcontender, 2019-12-01:
I updated the firmware on the controller when I first purchased it 2 weeks ago. Has there been any updates since? I can check again tomorrow when I have time.

FranticRain, 2019-12-01:
I wouldn’t know. I don’t own one. That seems like the only difference between the previous poster and you, though. I have no idea what else would be causing these to act differently.

newcontender, 2019-12-02:
Okay it seems it is working now after rebooting into my Windows 10 partition to check the firmware ( 4.8.1908.0.) When booting back into Catralina it was working fine... It seems it simply required a reboot after the initial installation reboot.

Either way, I'm happy to report everything is okay so far.

newcontender, 2019-12-03:
Playing wired is fine but Just to further understand the technicalities of all this. What would prevent the Xbox controller from being recognized in other applications via Bluetooth? I know it works with Steam. Is it simply a driver issue?

FranticRain, 2019-12-15:
It is a game issue. A lot of games are really dumb and do really dumb stuff like look for the driver specifically in order to connect with devices. Other games are only designed to work with USB based controllers. Long story short, macOS's controller support has been really bad for a really long time. And we can't take a Bluetooth device and make it look like a USB device. So the problem really is in the games, and there isn't much we can do about it.

newcontender, 2019-12-16:
Ah, I see. More specifically, I have been trying to get it to work with Openemu - a popular Multiplatform emulator. I haven't used any other Xbox controller with this application before. PS4 controller support via bluetooth has worked fine with this though.

I have done some digging and it seems this controller may just need a firmware update to rectify these issues. People have been complaining about bluetooth not working with their IOS devices and  such. I am not sure if you are familiar with the Openemu platform but have Xbox wireless controllers been able to work with them in the past?

FranticRain, 2019-12-16:
In my experience, OpenEmu has had some of the best controller support in an application that I’ve ever seen. It should support all controllers no mater how you connect them. Be it via the driver in wireless/wired or via Bluetooth. I originally made the driver because I wanted to play Metal Gear Solid via OpenEmu, so I can confirm that it works via all of the methods listed above. Personally, I always suggest that people use the best supported path, which is Bluetooth. Any issues specific to the Elite 2, I can’t comment on.

newcontender, 2019-12-16:
Funny, MGS was one of the first games I tested when I tried the Bluetooth on this controller. I will just bank on a firmware update resolving it in time. Thanks for your detailed explanations :)

mrhelmut, 2019-12-16:
Hi there!

We received the Elite Series 2 at the office and we're in that boat, it's not working on macOS (both wired and wireless, tested with the driver posted by @FranticRain earlier here).

Product ID: 0x0b05  (it's a different one!)
Vendor ID: 0x045e
Firmwave ID: 0x0903  (it's the latest, we updated it on Windows)

We're on Catalina.

Here's a packet sniffing dump: https://www.dropbox.com/s/uh5te32m3fuhog7/elite2?dl=1

Cheers!

FranticRain, 2019-12-16:
@newcontender You've probably already done this, but make sure to configure your controller in the OpenEmu preferences.

@mrhelmut You have the same PID as the OP, so this device should be recognized by the driver. What's the response in the preference pane when you connect the device? Can you confirm via the "enable driver" checkbox that the driver is set up correct and enables without error message? And your packet dump doesn't really tell me much. As far as it's concerned, you didn't press any buttons, and there was just a lot of useless noise from a device on the same bus. I don't think I need a packet sniff to solve your problem, though.

mrhelmut, 2019-12-16:
Ah right!

Enabling/disabling is fine.
When wired, it is found but absolutely nothing responds.
Over bluetooth it isn't found at all (while being connected to macOS properly).

Also, I noticed that the different PIDs are depending on the connection.

0x0b05 = bluetooth
0x0b00 = wired

I'll do another dump, this one comes from Windows VM which is likely the reason why the inputs didn't got through.

FranticRain, 2019-12-16:
Ah, perfect! Thanks for that PID find. That will help me clean up the plist.

Note that wired is the only thing this driver cares about, and that I can help you with. macOS supports this controller via Bluetooth natively, and therefore the driver is not used to control this device. Therefore, it doesn't appear in the preference pane. Not ideal, I realize. If you want to test it in Bluetooth mode, check out an HTML5 gamepad tester or something of that ilk. Any further support of Bluetooth Xbox One controllers is out of scope of this driver, currently.

For wired, I absolutely don't know what's going on. By all accounts, the packets that this device sends *should* work just fine. And they do, in fact, work for some people. This is beyond perplexing to me. Can you turn off the bluetooth on your device and then plug the controller in and see if it works?

mrhelmut, 2019-12-17:
I'm aware about this driver being for wired only, it's fine. :) Just thought mentioning my results.

macOS Catalina and iOS 13 don't support the Elite 2. It was advertised as supported at first, but it ended up not being supported and Apple removed it from the supported controllers list. It pairs but that's all.

Would it be possible that the driver conflicts with a previous version of it from override it with a driver with the same version number? (just a random thought since it appears they both are 1.0.0-alpha.5)

This is puzzling me too as those controllers should be using the same base protocol. I'm tempted to think it's an issue from our side.

FranticRain, 2019-12-17:
The installation process deletes all artifacts of all versions of previous installations before installing the new one. I do development builds with identical version numbers all the time and have no problems whatsoever. The version number is purely for customers and has no real bearing on the implementation or execution of the code.

My fear is that you are plugging the controller in, and it is pairing with the Bluetooth on your device. So you are recognized by the driver, but all buttons are being sent via Bluetooth. So if you could run that test I mentioned in my last comment where you turn off the Bluetooth on your device and plug in the controller, do you get a different result?

mrhelmut, 2019-12-17:
Just tested with the bluetooth completely turned off. Controller is recognized and appears in the list, but nothing more happens.

newcontender, 2019-12-17:
@mehelmut Just out of curiosity is the light on your Elite 2 home button solid when you test out the controller inputs? I noticed mine wouldn't register at one point unless the light was solid.

@FranticRain Yup, I checked the controller configuration in the OpenEmu preferences. It doesn't even appear when over Bluetooth but shows up fine when connected wired.

FranticRain, 2019-12-17:
The solid indicator light *should* confirm a proper pairing. There's a bad state that the controller can get in to in which the light would be solid, but we'd have to send a bad initialization packet to get that to happen. I just don't understand how both of you can have the same controller, with the same version of the driver, and have completely different results.

mrhelmut, 2019-12-18:
Home button is properly lit when wired and listed in the driver as "Xbox One Wireless Controller (Wired)" but nothing responds.

We tested multiple other controllers to make sure that our macOS installation isn't broken (X360, regular Xbox One, and Elite v1) and everything works.

I also checked if the mapping profiles inside the controller are fine and they are all set to default.

I'll send you a proper input dump later if that may prove useful (though I'm expecting there will be nothing new).

henrycc265, 2019-12-21:
god... so technical ... after reading all those comments I still don't have a clue what's happening .. so it the elite 2 controller working on macOS Catalina yet? I don't mind just wired support as soon as I can use it  to play deadcell

mrhelmut, 2019-12-21:
> so it the elite 2 controller working on macOS Catalina yet?

If you're speaking about Apple's official support for bluetooth controllers, no, the Elite Series 2 is not part of the official Catalina supported controllers (only the Xbox One S controller is supported; pre-2016 controllers aren't).

FranticRain, 2019-12-21:
@henrycc265 As for wired, I dunno, maybe. Try any of the builds posted in this thread and see if it works for you. Works for some people, doesn’t for others.

mariopavlov, 2019-12-23:
Hi,

I have tried the build from here on macOS Catalina with Xbox Elite Series 2. The controller is not recognised from "Xbox 360 Controllers". The controller is connected to the Mac via Bluetooth. Is there anything that I miss?

FranticRain, 2019-12-23:
@mariopavlov As I’ve explain in this thread before, and as stated in the README, this driver does nothing for Bluetooth devices. Those devices are supported natively and therefore do not appear in the preference pane. If you want to test a Bluetooth controller, use something like an HTML5 game pad tester.

grp, 2019-12-23:
@FranticRain I think some of the confusion is because this particular device doesn’t work over Bluetooth at all. Of course this driver wouldn’t change that.

newcontender, 2019-12-23:
@grp Bluetooth works on Steam for me, but that is all so far. I haven't had time to test it with many other games/applications.

FranticRain, 2019-12-23:
@grp That's fair. Ideally, I'd like to find all controllers in the preference pane and then disable settings customization for non-driver supported controllers and maybe call it `Xbox One Controller (Bluetooth) [Not Using 360Controller]`. But I'm not planning on putting too much more effort into how things currently are when I'm going to have to redo everything for DriverKit anyway. :)

Taconite12, 2020-01-07:
@FranticRain About to give my elite V2 a try later. Is this working now?

FranticRain, 2020-01-07:
@Taconite12 Nothing has changed. As stated a few posts up in this thread, you're welcome to try the build in this thread. It should work, and I have confidence that it does work. If you've used the controller via Bluetooth, make sure to disconnect it from the PC or phone you connected it to and plug it in to a Windows machine once to reset it before you plug it in to your Mac.

Taconite12, 2020-01-07:
@FranticRain What do you mean by reseting it on a windows machine? I am on vacation and do not have anything windows with me but hopefully it will work without reconnecting to an xbox. Downloading the build from this thread now and will report back.

Taconite12, 2020-01-07:
@FranticRain So it looks like it is recognized by the software just fine but there is no ability to test the controller and the controller seems to have no ability on the game (Fortnite) that i am playing.

FranticRain, 2020-01-08:
When I say "reset" you just need to plug it into a Windows machine that can properly start it up, and sort of "reset" it from Bluetooth state into wired state. I'm pretty sure that people having issues with this controller are experiencing this issue and not an actual issue with the Elite 2 specifically, seeing as how it works for some people.

jspiro, 2020-01-13:
I can confirm the build you provided in November worked out of the box for S2 after a restart. Any idea why it took like 5 real minutes for the installer script to run?

FranticRain, 2020-01-13:
Okay, that’s good. I’m going to basically declare the Elite V2 problem solved then. It seems like the rest of the issues are the same as #993. Going to leave this open since you’ll need a build in this thread to work, but I’m considering this done as soon as that other issue is fixed.

FranticRain, 2020-01-13:
@jspiro As to why install takes a long time, that’s all out of my hands. As far as I know, that’s just macOS introducing a random wait in order to drive you to distributing on the App Store. Also Obra Dinn is great. I hope you enjoy it.

[39 more replies]