360Controller/360Controller #967: Multiple Xbox One Wireless controller connection problem

honzasusek, 2019-11-07:
### Type of Controller
Xbox One

### OS Version
10.15

### Driver Version
1.0.0-alpha.5

### Connection Method
Wireless Adapter

### Device Name and Info
>Device Name:
XBOX ACC:

<br/>Product ID: 0x02fe
<br/>Vendor ID: 0x045e

>Device Name:
Controller

<br/>Product ID: 0x02ea
<br/>Vendor ID: 0x045e

### Response of Controller
First controller connects just fine (LED stays on, can play games). When I connect a second one, the first one will disconnect for a few second (blinking LED) leaving the second one connected, then the second controller will disconnect while the first one connects again and this switching goes on and on until I turn one of the controllers off or remove the adapter.

### Response in Preference Pane
New controller instances are added upon every connection, there's usually 20+ controllers in the panel after a minute or so.

### Detailed Explaination of Issue
When I connect more then one Xbox One controllers through the wireless adapter they keep alternating between connecting/disconnecting generating more and more controller instances every time.

medusalix, 2019-11-07:
Thanks for the feedback, I wasn't actually able to test the wireless dongle with multiple controllers myself because I only have one controller. I'm going to have to investigate the Wireshark dump where multiple controllers were paired and try to find the problem.

honzasusek, 2019-11-07:
@medusalix Sure, bring it on :)

medusalix, 2019-11-07:
@honzasusek Here you go: [WirelessOneController.kext.zip](https://github.com/360Controller/360Controller/files/3819947/WirelessOneController.kext.zip)

honzasusek, 2019-11-07:
@medusalix Now I can't even connect one. I did the following:

1. replaced the .kext with the new one under `/Library/Extensions`
2. restarted
3. adapter didn't light up, so I used `kextload` to load the extension manually (is it supposed to be like that?)
4. reconnected adapter which started responding, but the controller doesn't connect at all (both adapter & controller LEDs keep blinking, adapter gives up after a while, then the controller as well)

wired seems to work fine as it did before

medusalix, 2019-11-07:
@honzasusek Thanks for testing. Please try out this version: [WirelessOneController.kext.zip](https://github.com/360Controller/360Controller/files/3820501/WirelessOneController.kext.zip). I really need your logs to diagnose the problem (`Console.app` with filter `process:kernel`).

honzasusek, 2019-11-07:
@medusalix Tried out using the same process, results were the same. Here are my logs:

[logs.txt](https://github.com/360Controller/360Controller/files/3820613/logs.txt)

medusalix, 2019-11-09:
@honzasusek Please try this one: [WirelessOneController.kext.zip](https://github.com/360Controller/360Controller/files/3827799/WirelessOneController.kext.zip). It works with my controller and should (in theory) support multiple controllers.

honzasusek, 2019-11-10:
@medusalix All good now :) My mistake though, I was loading the extension through `kextload` instead of `kextutil` as indicated in the readme. Sorry about that, hope it didn't cause much extra work.