360Controller/360Controller #968: Failed to boot XBox 360 Controller on catalina 10.15.1

suesunss, 2019-11-08:
### Type of Controller
Replace this text with: Original Xbox, Xbox 360, Xbox One, All or N/A

xbox 360

### OS Version
Replace this text with the version of macOS that you are running

macos catalina version 10.15.1 (19B88)

### Driver Version
Replate this text with the version or versions of the driver that you are using or have tested
Version 1.0.0-alpha.5

### Connection Method
Replace this text with: Wired, Wireless Adapter, Bluetooth, N/A or more specific method if applicable

wired 

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID: 0x028e
<br/>Vendor ID: 0x045e

### Response of Controller
Replace this text with information on how the physical controller reacts. Does the guide button light up? Does it vibrate? Etc.

>> Nothing works. I can find 'Controller' from System Information, but the controller doesn't work.

### Response in Preference Pane
Replace this text with information about the controller's reaction in the preference pane. Does the controller appear in the preference pane? Do all of the buttons, joysticks, and triggers act as expected?

>> Controller appear correctly in the preference pane. All the buttons, joysticks and triggers dont't work.

### Detailed Explaination of Issue

>> I just updated my Mac (2015 mid, 15 inch, retina) to the latest catalina MacOS, I want to drive my XBox 360 Controller, I am using XBox 360 Controller Driver (Version 1.0.0-alpha.5). All the installation process works fine, but the controller cannot be drived correctly, the central button doesn't bright, none of the buttons work.

FranticRain, 2019-11-08:
Is this an official Microsoft controller or is this from some third party?

suesunss, 2019-11-09:
> Is this an official Microsoft controller or is this from some third party?

This is an official MS controller.

FranticRain, 2019-11-10:
Just to be sure, because the driver appears to be loaded correctly, does this controller work with a Windows PC or an Xbox console?

suesunss, 2019-11-11:
@FranticRain yes, the controller works fine with a windows PC, I haven't tested with an Xbox console. By the way, this is the link to the precise controller I bought. 
https://www.amazon.fr/gp/product/B004JU0JSK

Bleysus, 2019-11-18:
Same issue. The given link worked for me, thanks! 

macOS 10.15.1 (19B88)
Product ID: 0x028e
Vendor ID: 0x045e  (Microsoft Corporation)