360Controller/360Controller #969: Cannot 'Enable driver'

Callum1414, 2019-11-09:
Xbox One

OSX El Capitan

Version 1.0.0-alpha.5
Wireless (for some reason I can't get the wired connection to work)
MacBook Pro (13-inch, Mid 2012)

the controller can wirelessly connect and I can use it to play some games but the Xbox 360 driver says "no controller detected" even though the controller is connected

I cannot enable the "enable driver" box and if I try to click it this pop-up comes up:

Cannot load driver
The driver has failed to load due to an unknown error, code: -603946986. Please report this error to the driver maintainer.

pls help because I'd love to use this software so that I can play other games and change keybinds

Callum1414, 2019-11-09:
I installed the alpha 3 version and now it works

FranticRain, 2019-11-09:
This is a known issue with versions of macOS pre Mojave. Here's a version of the driver built against the same codebase as alpha 5 (has all of the new updates) that will work on your machine:
https://github.com/360Controller/360Controller/files/3812645/360ControllerInstall.dmg.zip

The reason the controller doesn't show up in the preference pane when you are using it "wireless" is because you are using it via Bluetooth. macOS supports Xbox controllers natively via Bluetooth. Therefore, the driver is not involved, and does not detect controllers that are not using the driver.

closed 2019-11-15 by FranticRain

sauravchangde, 2024-04-26:
I'm facing the same issue.
Mac Details: Air M2