360Controller/360Controller #970: Wired 360 controller

JoelGG, 2019-11-10:
### Type of Controller
Xbox 360

### OS Version
10.14.6

### Driver Version
360Controller 1.0.0-alpha.2

### Connection Method
Wired

### Device Name and Info
Device Name: Controller
Product ID: 0x028e
Vendor ID: 0x045e

### Response of Controller
Guide button repeatedly flashes.

### Response in Preference Pane
Controller does not appear in preferences pane.

### Detailed Explaination of Issue
Controller is not recognised in the preference pane, and therefor does not work with any games.

FranticRain, 2019-11-10:
Use the most recent version of the driver (alpha 5) and let me know if that fixes your issue.

JoelGG, 2019-11-13:
That fixed it thanks.

closed 2019-11-13 by JoelGG