360Controller/360Controller #974: Wireless Xbox Controller not identified

kstevens14, 2019-11-17:
### Type of Controller
wireless xbox controller model 1708

### OS Version
Catalina 10.15.1

### Driver Version
ive tried multiple versions of 360 Controller and none of them have worked. Currently using Alpha 4

### Connection Method
ive tried wireless and wired connections

### Device Name and Info
MacBook Pro

### Response of Controller
The controller when plugged in will vibrate but no light will come on and it will still say no devices found. Ive connected the controller to bluetooth and it will say connected but the buttons are unresponsive.

### Response in Preference Pane
Controller won't show up at all.

### Detailed Explaination of Issue
The controller will not show up when selecting a device. ive tried many different versions and restarted my computer and nothing has worked. Ive tried older and newer versions of 360Controller. Not sure if I am doing something wrong or if it is not compatible with my computer.

Any help is appreciated.

FranticRain, 2019-11-17:
You must update your original post with your controller information (it's what's supposed to go into "device name and info"). A guide on how to do so can be found here: https://github.com/360Controller/360Controller#how-do-i-find-my-vendor-id-and-product-id.

Controller information is only not required for issue that do not involve controllers. For example, a feature request that isn't specific to any controller, or maybe a typo somewhere in the code or documentation. For all issues individuals have with getting a controller working, controller information is ABSOLUTELY VITAL to solving your problem. Not only does it let us know if your controller is already supported, it also lets us know that the operating system can see the controller and is a crucial step in resolving your issue.

Please notify my when you have updated your original post with your controller information, or let me know if you are unable to find it.

kstevens14, 2019-11-17:
@FranticRain I have updated the post information

kstevens14, 2019-11-17:
@FranticRain the xbox controller won't show up under USB devices even though it is plugged in and vibrated. not sure what that means. Got the product and vendor IDs from when I connected in through bluetooth

FranticRain, 2019-11-17:
A couple things:

1. The reason it isn't showing up in the USB devices is because you are using what is known as a "charge" cable to connect it. That cable only has the connections to provide power to the controller. It doesn't contain the connections required to send data back and forth the controller. The controller can't work with a charge cable. Try again with a different cable.

2. The driver doesn't work with Bluetooth connections. The controller works natively though macOS when used via Bluetooth. You won't see inputs in the preference pane. If you want to see the controller working in Bluetooth mode, I recommend an HTML gamepad tester like this one: https://html5gamepad.com/

Let me know if you have any further questions.

kstevens14, 2019-11-17:
@FranticRain got it to work, thanks for your help

closed 2019-11-17 by kstevens14