360Controller/360Controller #975: I never can get the 360 controller gets work

frognum, 2019-11-17:
### Type of Controller
Wireless MS Xbox 360 controller with and without charging cable

### OS Version
Catalina

### Driver Version
alpha 5

### Connection Method
Wireless (with MS adaptor) and charging cable.

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID:
<br/>Vendor ID:

### Response of Controller
The controller do nothing. The cpanel options shows "No device found"

### Response in Preference Pane
Replace this text with information about the controller's reaction in the preference pane. Does the controller appear in the preference pane? Do all of the buttons, joysticks, and triggers act as expected?

### Detailed Explaination of Issue
It doesn't work

FranticRain, 2019-11-17:
A couple things:
1. The Xbox 360 controller is not like the Xbox One controller. Connecting a wireless Xbox 360 controller to a computer via a cable does NOT make it a wired controller. Xbox 360 Wireless controllers are always wireless and that Play N Charge cable will NEVER work.
2. Wireless adapters are currently bugged in macOS Catalina. This is a known bug that is being worked on.

FranticRain, 2019-11-22:
To follow progress on wireless adapter fixes, follow #955

closed 2019-11-22 by FranticRain