360Controller/360Controller #976: New pdp wired controller mac 10.15.1 not detected

Dzaabir, 2019-11-19:
### Type of Controller

Pdp wired controller for xbox one & windows

### OS Version

Mac os catalina 10.15.1

### Driver Version

I install the 1.0.0-alpha.5 and i follow this link to install this other version : https://github.com/360Controller/360Controller/files/2848523/Install360Controller.pkg.zip

This both versions doesn't works for me

### Connection Methode 
Wired usb

### Device Name and Info

>Device Name: PDP wired controller for xbox one - royal purple
<br/>Product ID: 0x02bd
<br/>Vendor ID: 0x0e6f

Version :	1.10
Numéro de série :	0000BD5FEDA67616
Vitesse :	Jusqu’à 12 Mb/s
Fabricant :	Performance Designed Products
Identifiant de l’emplacement :	0x14300000 / 8
Courant disponible (mA) :	500
Exploitation supplémentaire actuelle (mA) :	0

### Response of Controller

no devices found, nothing happens when i press the xbox button

### Response in Preference Pane

no devices found, no lights when i press the xbox button, nothing happens

### Detailed Explaination of Issue

I try this device on my windows partition and it works, i can use it in fortnite for example.
But on my mac with this  2 drivers I can't use it.
I hope you will help me, i thank you in advance for your answer.

Antony.

r00t-us3r, 2019-11-21:
Also experiencing the same issue with PDP controller macOS Catilana

FranticRain, 2019-11-22:
@r00t-us3r Add your product and vendor IDs to this thread.

FrostbourneFlame, 2019-12-10:
Sorry for the late notice on this rather deserted issue, but I'm also experiencing the same problems, also with macOS Catalina. I don't know the driver version for me since I deleted the initial installation package, but whenever I try and download the latest release (1.0.0-alpha.5), my security & privacy says that "it can't be opened because Apple cannot check it for malicious software."

Sorry for going a bit off topic on that, but my product is a "**Royal Purple PDP Wired Controller for Xbox One and Windows**". 

Device Info/Vendor IDs:

  Product ID:	0x02bd
  Vendor ID:	0x0e6f
  Version:	1.10
  Serial Number:	00007ECB72ED0281
  Speed:	Up to 12 Mb/s
  Manufacturer:	Performance Designed Products
  Location ID:	0x14200000 / 29
  Current Available (mA):	500
  Extra Operating Current (mA):	0

FranticRain, 2019-12-22:
This device should be included in this version of the driver.

[360ControllerInstall.dmg.zip](https://github.com/360Controller/360Controller/files/3993108/360ControllerInstall.dmg.zip)

closed 2019-12-22 by FranticRain

SaintsConnor, 2022-12-27:
Still not working, no lights nothing. Controller: PFP Ion White Wired