360Controller/360Controller #977: Sven  gc-3050

yaBliznyk, 2019-11-21:
### Type of Controller
Sven  gc-3050 on xbox 360 for windows mode

### OS Version
Catalina 10.15.1

### Driver Version
0.16.11

### Connection Method
Bluetooth adapter. Bluetooth on macbook doesn't see this controller

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: XBOX 360 For Windows
<br/>Product ID: 0x028e
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller
No devices found, but all indicators on controller are working

### Response in Preference Pane
No devices found

### Detailed Explaination of Issue
I try to connect by usb cable, but it just start charging

FranticRain, 2019-11-21:
We don't support controllers that do not work with Xbox hardware. We also do not support controllers that use a Bluetooth connection. As such, its support is outside the scope of this driver. We cannot and will not support this controller.

closed 2019-11-21 by FranticRain

yaBliznyk, 2019-11-21:
Thanks for answer!