360Controller/360Controller #979: PDP Wired Controller for Xbox One - Stealth Series | Phantom Black, Catalina 10.15.1

kantagara, 2019-11-22:
### Type of Controller
PDP Wired Controller for Xbox One - Stealth Series | Phantom Black:

### OS Version
10.15.1 Catalina
### Driver Version
Latest Alpha

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID:0x02a4
<br/>Vendor ID:0x0e6f

### Response of Controller
Replace this text with information on how the physical controller reacts. Does the guide button light up? Does it vibrate? Etc.
Nothing happens although in USB

### Response in Preference Pane
Replace this text with information about the controller's reaction in the preference pane. Does the controller appear in the preference pane? Do all of the buttons, joysticks, and triggers act as expected?
No it says it's not detected

### Detailed Explaination of Issue
I can see this name: PDP Wired Controller for Xbox One - Stealth Series | Phantom Black: in the System Information, I've installed the drivers but nothing seems to work

FranticRain, 2019-11-23:
Can you install this version of the driver and let me know if it solves your problem?[360ControllerInstall.dmg.zip](https://github.com/360Controller/360Controller/files/3882083/360ControllerInstall.dmg.zip)

kantagara, 2019-11-23:
Hey @FranticRain , unfortunately, the issue is the same (it's detected in System Perferences but the movement isn't detected). I'm actually on Catalina 10.15.2 developers beta, totally forgot that I've updated to it. Also, is it worth mentioning that I'm on a Hackintosh?

kantagara, 2019-11-23:
Hm Tried it now on real Macbook Pro, and some message popped up that I need to open security & privacy to allow something, but when I opened it I couldn't find anything related to this.

FranticRain, 2019-11-23:
So there's an issue with macOS versions after 10.11 that cause these controllers to not function properly. It seems to a bug in Apple's code, as these sorts of controllers worked before 10.11. I've been trying to find ways to fix it, but haven't managed to find a solution yet.

As for the real macbook pro issue, go into the preference pane, go to the "advanced" tab and check the "enable driver" checkbox. It will take you through a debugging process that should help you figure out what went wrong.

MANTENN, 2021-05-31:
Its detected under System Report:
```
PDP Wired Controller for Xbox One - Stealth Series | Phantom Black:

  Product ID:	0x02a4
  Vendor ID:	0x0e6f
  Version:	1.0f
  Serial Number:	00001B8F57684A08
  Speed:	Up to 12 Mb/s
  Manufacturer:	Performance Designed Products
  Location ID:	0x14340000 / 16
  Current Available (mA):	500
  Current Required (mA):	500
  Extra Operating Current (mA):	0
```

It's not recognized by the driver. Not sure why.

MANTENN, 2021-05-31:
```
default	22:50:12.875205-0700	kernel	PDP Wired Controller for Xbox One - Stealth Series | Phantom Black: family specific matching fails
default	22:50:12.875209-0700	kernel	PDP Wired Controller for Xbox One - Stealth Series | Phantom Black: family specific matching fails
default	22:50:12.875231-0700	kernel	Xbox360Peripheral::probe(PDP Wired Controller for Xbox One - Stealth Series | Phantom Black)
default	22:50:12.875235-0700	kernel	Xbox360Peripheral::start(PDP Wired Controller for Xbox One - Stealth Series | Phantom Black) <1>
default	22:50:12.877273-0700	kernel	Unknown device release 010f
default	22:50:12.906049-0700	kernel	start - unable to find chatpad interface
default	22:50:13.050975-0700	kernel	read - failed to start (0xe00002c2)
default	22:50:13.089555-0700	kernel	Xbox360Peripheral::start(PDP Wired Controller for Xbox One - Stealth Series | Phantom Black) <1> failed
default	22:50:49.213395-0700	useractivityd	[PBOARD CONTROLLER] Got type update from client, gen: 3, <private>
default	22:50:49.266453-0700	useractivityd	[PBOARD CONTROLLER] Got type update from client, gen: 3, <private>
default	22:52:00.528516-0700	useractivityd	[PBOARD CONTROLLER] Got type update from client, gen: 4, <private>
default	22:52:00.579197-0700	useractivityd	[PBOARD CONTROLLER] Got type update from client, gen: 4, <private>
default	22:53:04.161950-0700	useractivityd	[PBOARD CONTROLLER] Got type update from client, gen: 5, <private>
default	22:53:04.213462-0700	useractivityd	[PBOARD CONTROLLER] Got type update from client, gen: 5, <private>
default	22:53:04.401437-0700	useractivityd	[PBOARD CONTROLLER] Got type update from client, gen: 6, <private>
default	22:53:04.452060-0700	useractivityd	[PBOARD CONTROLLER] Got type update from client, gen: 6, <private>
```

MANTENN, 2021-05-31:
Seems related to: https://github.com/360Controller/360Controller/issues/942

MANTENN, 2021-06-01:
> It's not recognized by the driver. Not sure why.

It gets recognized by the driver but not in the system preferences.

MANTENN, 2021-06-09:
Fixed in the Alpha release.