360Controller/360Controller #980: Wired Xbox one controller not showing up

mcauley99, 2019-11-23:
### Type of Controller
Xbox One

### OS Version
10.11

### Driver Version
Tried the 5th one and fourth with no success. Currently have 4th installed. 

### Connection Method
Wired micro usb

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:iMac
<br/>Product ID:0x02ea
<br/>Vendor ID:0x045e

### Response of Controller
Vibrates when plugged in. No light. 

### Response in Preference Pane
Doesn’t appear in the preferences. Says no devices found. Item1 sometimes appears for a split second before it reverts to no devices found. It is a secured connection I have used this exact controller and micro usb on my windows pc. 

### Detailed Explaination of Issue
Controller simply doesn’t show up. I’ve tried restarting and uninstalling my drivers. I’ve tried plugging the controller after I’ve opened the pane and before.

FranticRain, 2019-11-23:
Can you run this command in your terminal? It will test whether or not the driver is loaded as expected. It will ask for your password as it needs administrator privileges. Let me know what it says.
```
sudo kextutil /Library/Extensions/360Controller.kext
```

gl2eyalien, 2019-11-27:
I am encountering the same issue.  Here is the response to the command you posted:

sudo kextutil /Library/Extensions/360Controller.kext
Untrusted kexts are not allowed
Kext with invalid signature (-67062) denied: /Library/StagedExtensions/Library/Extensions/9217F9CA-675B-480D-AAE2-A2B595ABD43E.kext
Bundle (/Library/Extensions/daspi.kext) failed to validate, deleting: /Library/StagedExtensions/Library/Extensions/9217F9CA-675B-480D-AAE2-A2B595ABD43E.kext
Unable to stage kext (/Library/Extensions/daspi.kext) to secure location.
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

(kernel) Notice - new kext com.apple.driver.usb.AppleUSBHostPlatformProperties, v1.2 matches prelinked kext but can't determine if executables are the same (no UUIDs).

FranticRain, 2019-11-28:
@gl2eyalien You haven’t approved the driver in your system preferences. Whenever you connect a controller or run that command, an option should should up in the security section of the preferences that will let you allow the driver to run.

TxmyXV, 2019-12-13:
mine shows i have a macbook pro 8.1 late 2011 with sierra 10.12.6 and i installed the driver version v1.0.0-alpha.4
 Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

(kernel) kxld[com.mice.driver.Xbox360Controller]: The super class vtable '__ZTV9IOService' for vtable '__ZTV17Xbox360Peripheral' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The super class vtable '__ZTV11IOHIDDevice' for vtable '__ZTV20ChatPadKeyboardClass' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The super class vtable '__ZTV11IOHIDDevice' for vtable '__ZTV22Xbox360ControllerClass' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV22Xbox360Pretend360Class was not patched because its parent, the __ZTV22Xbox360ControllerClass, was not patchable.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV22XboxOneControllerClass was not patched because its parent, the __ZTV22Xbox360ControllerClass, was not patchable.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV22XboxOnePretend360Class was not patched because its parent, the __ZTV22XboxOneControllerClass, was not patchable.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV27XboxOriginalControllerClass was not patched because its parent, the __ZTV22Xbox360ControllerClass, was not patchable.
(kernel) Can't load kext com.mice.driver.Xbox360Controller - link failed.
(kernel) Failed to load executable for kext com.mice.driver.Xbox360Controller.
(kernel) Kext com.mice.driver.Xbox360Controller failed to load (0xdc008016).
(kernel) Failed to load kext com.mice.driver.Xbox360Controller (error 0xdc008016).
Failed to load /Library/Extensions/360Controller.kext - (libkern/kext) link error.
Check library declarations for your kext with kextlibs(8).

TxmyXV, 2019-12-13:
and if i check "enable driver" in the advanced tab it gives me this error  "The driver has failed to load due to an unknown error, code: -603946986."

FranticRain, 2019-12-15:
@TxmyXV Use alpha 5. It fixes that bug.

This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-12-15 by FranticRain