360Controller/360Controller #981: Xbox 360 Wired Controller no recognised

CelestialRiot, 2019-11-23:
Third party Xbox 360 Controller

OS: macOS Sierra

Version: 360Controller 1.0.0-alpha.5

Wired Connection

>Device Name: Gamepad for Xbox 360
<br/>Product ID: 0x0401
<br/>Vendor ID: 0x0e6f

### Response of Controller
All 4 sections of the guide light green when it is inserted into the USB port, no other responses otherwise

### Response in Preference Pane
"No devices found"

### Detailed Explaination of Issue
When the controller is connected, it is not recognised and so "it doesn't work".

FranticRain, 2019-11-23:
Can you go to the "advanced" tab of the preference pane for the driver and check the "enable driver" checkbox? It should take you through a debugging process that should help you resolve this issue.

CelestialRiot, 2019-11-23:
I downloaded the newer version from one of the threads and it works properly now and responds well in https://html5gamepad.com/

closed 2019-11-23 by CelestialRiot

FranticRain, 2019-11-24:
Can you tell me which thread you got it from? I try a bunch of different things and I don’t always know which ones work and which ones don’t.