360Controller/360Controller #983: Xbox one S controller connects and lights up when wired, but not recoginzing button inputs. Nothing with bluetooth.

Gladiatorbbn, 2019-12-02:
### Type of Controller
Xbox One

### OS Version
10.15.1

### Driver Version
Version 1.0.0-alpha.5

### Connection Method
Wired, Bluetooth

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID: 0x9cb1
<br/>Vendor ID: 0x8086

### Response of Controller
When connected by micro-USB, the controller vibrates and the guide button lights up. When connected by bluetooth, nothing.

### Response in Preference Pane
When connected by wired, it recognizes a wired Xbox one controller. When connected by bluetooth, no recognition. While connected via wire, there is nothing happening when the sticks, triggers, or buttons are pushed and moved.

### Detailed Explaination of Issue
Been wanting to use an Xbox one S controller to play Stadia. The left stick, "A" button, and "B" button work, but the other buttons, sticks, and triggers are a mess. I am unable to remap, as there is no response when I follow the instruction to press a button. The first instruction is to press the up button on the d-pad. When I do so, there is no response. This is all happening while wired. While connected to bluetooth, I receive "No devices found". I have additionally tried "Pretend to be an Xbox 360 Controller", but the same issues persist.

FranticRain, 2019-12-15:
Okiedoke, a couple things of note here.

First, just use the controller via Bluetooth. It works with macOS natively, don't use this driver unless you have to. This means that Bluetooth connections will not be recognized by the preference pane. You'll need to test it with something like a web-based gamepad tester to see if it's working. [Something like this.](https://html5gamepad.com/) Feel free to just follow these instructions, uninstall the driver, and ignore the rest of this message.

Second, via wired, your issue seems to claim that none of the buttons, triggers, or sticks are recognized in the preference pane, but Stadia recognizes A, B, and the left stick? I just want to be sure, what all works and doesn't work in the preference pane when the controller is connected via a wired connection?

Gladiatorbbn, 2019-12-15:
Thank you for your response. Firstly, I was able to find a fix by connecting the controller to the SteamLink app on my phone and mapping the buttons. The button mapping layout learned from the app was retained and identical when connected to my Mac via bluetooth. 
Additionally to answer your question, before the controller was mapped correctly and was connected via wire, the only correct inputs were: A, B, and left stick. Some of the other buttons had no response. The view and menu buttons were LB and RB, respectively. This was the case when the controller was connected via wire.

closed 2019-12-15 by Gladiatorbbn