360Controller/360Controller #984: Kernel panic with wireless adapter

stefanfisk, 2019-12-09:
### Type of Controller
Xbox 360

### OS Version
10.14.6

### Driver Version
1.0 alpha 5

### Connection Method
Wireless Adapter

### Device Name and Info
Device Name:
Product ID: 0x0719
Vendor ID: 0x045e

### Response of Controller
Everything was working great until the computer went to sleep, and after that the controller was no longer recognised.

### Response in Preference Pane
The controller appeared and worked as expected until the computer went to sleep, and after that the controller was not longer visible in the pref pane.

### Detailed Explaination of Issue
After a fresh install of 1.0 alpha 5 followed by a reboot the controller was working perfectly until the computer went to sleep. After waking from sleep, the pref pane no longer recognised the controller. I then disconnected the wireless adapter, but it was still visible in the device tree even after waiting about a minute and refreshing multiple times. When I then connected the adapter to the other USB port (this is on a MBP 15 2014) the adapter appear twice in the device tree, but the controller was still not recognised by the pref pane. After that I left the computer, and when I came back and tried to wake it, it required a power button press to start, at which point it did a full boot.

After this I have used the adapter successfully, but I do not believe that I have tried letting the computer sleep with it connected, and due to work deadlines I'm currently not risking any more panics until I know that I have time to sort things out if something goes really wrong.

Below you can find the crash report.

```
Anonymous UUID:       D6DC38AB-54CA-369F-720D-9B56B891D0DA

Sun Dec  8 19:11:42 2019

*** Panic Report ***
panic(cpu 0 caller 0xffffff801b6dbadd): Kernel trap at 0xffffff801bc35e77, type 13=general protection, registers:
CR0: 0x000000008001003b, CR2: 0x0000600003ce0000, CR3: 0x000000001fc81000, CR4: 0x00000000001626e0
RAX: 0xc0ffee1aaa4d5554, RBX: 0xffffff8060070810, RCX: 0xffffff80566c8600, RDX: 0xffffff803ca9ad20
RSP: 0xffffff81fb08be10, RBP: 0xffffff81fb08be10, RSI: 0x0000000000000000, RDI: 0xffffff8060070810
R8:  0x0000000000000000, R9:  0x0000000000000008, R10: 0xffffff801bea0700, R11: 0xffffff81fa91d000
R12: 0xffffff80566c8650, R13: 0xffffff80566c8658, R14: 0xffffff803ca9ad20, R15: 0xffffff80566c8600
RFL: 0x0000000000010246, RIP: 0xffffff801bc35e77, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000600003ce0000, Error code: 0x0000000000000000, Fault CPU: 0x0, PL: 0, VF: 0

Backtrace (CPU 0), Frame : Return Address
0xffffff801b351290 : 0xffffff801b5ae6ed 
0xffffff801b3512e0 : 0xffffff801b6ea185 
0xffffff801b351320 : 0xffffff801b6db8ba 
0xffffff801b351390 : 0xffffff801b55bb40 
0xffffff801b3513b0 : 0xffffff801b5ae107 
0xffffff801b3514d0 : 0xffffff801b5adf53 
0xffffff801b351540 : 0xffffff801b6dbadd 
0xffffff801b3516b0 : 0xffffff801b55bb40 
0xffffff801b3516d0 : 0xffffff801bc35e77 
0xffffff81fb08be10 : 0xffffff7f9bf19830 
0xffffff81fb08be30 : 0xffffff801bc2c49a 
0xffffff81fb08be80 : 0xffffff801bc2f5ab 
0xffffff81fb08bf00 : 0xffffff801bc2f337 
0xffffff81fb08bf50 : 0xffffff801bc30e26 
0xffffff81fb08bfa0 : 0xffffff801b55b0ce 
      Kernel Extensions in backtrace:
         com.apple.iokit.IOHIDFamily(2.0)[4D861DA4-FE3A-3D9D-8123-038E9FBFE182]@0xffffff7f9bf15000->0xffffff7f9bf93fff

BSD process name corresponding to current thread: kernel_task

Mac OS version:
18G95

Kernel version:
[... 154 lines cut ...]
com.apple.driver.AppleACPIPlatform	6.1
com.apple.driver.AppleSMC	3.1.9
com.apple.iokit.IOPCIFamily	2.9
com.apple.iokit.IOACPIFamily	1.4
com.apple.kec.pthread	1
com.apple.kec.Libm	1
com.apple.kec.corecrypto	1.0

EOF
Model: MacBookPro11,3, BootROM 156.0.0.0.0, 4 processors, Intel Core i7, 2,5 GHz, 16 GB, SMC 2.19f12
Graphics: kHW_IntelIrisProItem, Intel Iris Pro, spdisplays_builtin
Graphics: kHW_NVidiaGeForceGTX750MItem, NVIDIA GeForce GT 750M, spdisplays_pcie_device, 2 GB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x134), Broadcom BCM43xx 1.0 (7.77.61.2 AirPortDriverBrcmNIC-1305.8)
Bluetooth: Version 6.0.14d3, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0512F, 500,28 GB
USB Device: USB 3.0 Bus
USB Device: Internal Memory Card Reader
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.1
```

stefanfisk, 2019-12-10:
Just about the same thing happened today. After using the controller without any issues, I disconnected the adapter and put the computer away. Now a couple of hours later when I tried waking it it had panicked.

```
Anonymous UUID:       D6DC38AB-54CA-369F-720D-9B56B891D0DA

Sun Dec  8 19:11:42 2019

*** Panic Report ***
panic(cpu 0 caller 0xffffff801b6dbadd): Kernel trap at 0xffffff801bc35e77, type 13=general protection, registers:
CR0: 0x000000008001003b, CR2: 0x0000600003ce0000, CR3: 0x000000001fc81000, CR4: 0x00000000001626e0
RAX: 0xc0ffee1aaa4d5554, RBX: 0xffffff8060070810, RCX: 0xffffff80566c8600, RDX: 0xffffff803ca9ad20
RSP: 0xffffff81fb08be10, RBP: 0xffffff81fb08be10, RSI: 0x0000000000000000, RDI: 0xffffff8060070810
R8:  0x0000000000000000, R9:  0x0000000000000008, R10: 0xffffff801bea0700, R11: 0xffffff81fa91d000
R12: 0xffffff80566c8650, R13: 0xffffff80566c8658, R14: 0xffffff803ca9ad20, R15: 0xffffff80566c8600
RFL: 0x0000000000010246, RIP: 0xffffff801bc35e77, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000600003ce0000, Error code: 0x0000000000000000, Fault CPU: 0x0, PL: 0, VF: 0

Backtrace (CPU 0), Frame : Return Address
0xffffff801b351290 : 0xffffff801b5ae6ed 
0xffffff801b3512e0 : 0xffffff801b6ea185 
0xffffff801b351320 : 0xffffff801b6db8ba 
0xffffff801b351390 : 0xffffff801b55bb40 
0xffffff801b3513b0 : 0xffffff801b5ae107 
0xffffff801b3514d0 : 0xffffff801b5adf53 
0xffffff801b351540 : 0xffffff801b6dbadd 
0xffffff801b3516b0 : 0xffffff801b55bb40 
0xffffff801b3516d0 : 0xffffff801bc35e77 
0xffffff81fb08be10 : 0xffffff7f9bf19830 
0xffffff81fb08be30 : 0xffffff801bc2c49a 
0xffffff81fb08be80 : 0xffffff801bc2f5ab 
0xffffff81fb08bf00 : 0xffffff801bc2f337 
0xffffff81fb08bf50 : 0xffffff801bc30e26 
0xffffff81fb08bfa0 : 0xffffff801b55b0ce 
      Kernel Extensions in backtrace:
         com.apple.iokit.IOHIDFamily(2.0)[4D861DA4-FE3A-3D9D-8123-038E9FBFE182]@0xffffff7f9bf15000->0xffffff7f9bf93fff

BSD process name corresponding to current thread: kernel_task

Mac OS version:
18G95

Kernel version:
Darwin Kernel Version 18.7.0: Tue Aug 20 16:57:14 PDT 2019; root:xnu-4903.271.2~2/RELEASE_X86_64
Kernel UUID: C41337A1-0EC3-3896-A954-A1F85E849D53
Kernel slide:     0x000000001b200000
Kernel text base: 0xffffff801b400000
__HIB  text base: 0xffffff801b300000
System model name: MacBookPro11,3 (Mac-2BD1B31983FE1663)

System uptime in nanoseconds: 5839203558476
last loaded kext at 5043949757873: com.mice.driver.Xbox360Controller	1.0 (addr 0xffffff7f9f42f000, size 90112)
last unloaded kext at 5028718488566: com.mice.driver.Xbox360Controller	1.0 (addr 0xffffff7f9f42f000, size 49152)
loaded kexts:
com.mice.driver.Xbox360Controller	1.0
org.virtualbox.kext.VBoxNetAdp	6.0.14
org.virtualbox.kext.VBoxNetFlt	6.0.14
org.virtualbox.kext.VBoxUSB	6.0.14
com.intel.kext.intelhaxm	6.2.1
org.virtualbox.kext.VBoxDrv	6.0.14
com.mice.driver.Wireless360Controller	1.0
com.mice.driver.WirelessGamingReceiver	1.0
com.intel.driver.EnergyDriver	3.6.1
com.apple.driver.AudioAUUC	1.70
com.apple.fileutil	20.036.15
com.apple.driver.AppleHWSensor	1.9.5d0
com.apple.filesystems.autofs	3.0
com.apple.driver.AGPM	110.25.11
com.apple.driver.ApplePlatformEnabler	2.7.0d0
com.apple.driver.X86PlatformShim	1.0.0
com.apple.driver.AppleGraphicsDevicePolicy	3.50.12
[... 126 lines cut ...]
com.apple.driver.AppleACPIPlatform	6.1
com.apple.driver.AppleSMC	3.1.9
com.apple.iokit.IOPCIFamily	2.9
com.apple.iokit.IOACPIFamily	1.4
com.apple.kec.pthread	1
com.apple.kec.Libm	1
com.apple.kec.corecrypto	1.0

EOF
Model: MacBookPro11,3, BootROM 156.0.0.0.0, 4 processors, Intel Core i7, 2,5 GHz, 16 GB, SMC 2.19f12
Graphics: kHW_IntelIrisProItem, Intel Iris Pro, spdisplays_builtin
Graphics: kHW_NVidiaGeForceGTX750MItem, NVIDIA GeForce GT 750M, spdisplays_pcie_device, 2 GB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x134), Broadcom BCM43xx 1.0 (7.77.61.2 AirPortDriverBrcmNIC-1305.8)
Bluetooth: Version 6.0.14d3, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0512F, 500,28 GB
USB Device: USB 3.0 Bus
USB Device: Internal Memory Card Reader
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.1
```

paweldziura, 2020-04-15:
Same here. Actually it happens very often when I sleep my computer. Later on when I want to wake it up, I see that the system crashed and it boots the system up.