360Controller/360Controller #985: roller.kext is already open and has a different CFBundleIdentifier from identifier->path cache entry 162.

electronstudio, 2019-12-10:
### Type of Controller
Xbox One

### OS Version
10.14.6

### Driver Version
 1.0.0-alpha.5

### Connection Method
Wired (USB-C)

### Device Name and Info
Device Name: xbox one elite controller series 2
<br/>Product ID: 0x0b00
<br/>Vendor ID: 0x045e

### Response of Controller
Sometimes vibrates when connected.  White light stays on.

### Response in Preference Pane
No devices found.

### Detailed Explaination of Issue
Have checked Security & Privacy, there is no option to allow there.

Here is output from kextutil:

```
> sudo kextutil /Library/Extensions/360Controller.kext
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Can't open CFBundle for /System/Library/Extensions/roller.kext.
Can't create kext from cache: /System/Library/Extensions/roller.kext is already open and has a different CFBundleIdentifier from identifier->path cache entry 162.
Warnings:
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done):
        ChatPadKeyboardEvents

Can't create kext from cache: /System/Library/Extensions/roller.kext is already open and has a different CFBundleIdentifier from identifier->path cache entry 162.
```

FranticRain, 2019-12-15:
From a fresh reboot, before connecting ANY devices to the system, can you run that command and see if you get the same result? Best guess is that you've somehow paired to a device, and your attempt to load it again is just letting you know that the kext is already loaded.

Also, if you could provide the output of this command whenever you see this specific error, that'd be great. That will let me know what is loaded in its place.
```
kextstat | grep 360Controller
```

And third, there is a special version of the driver in a different thread for the Elite Series 2. You'll probably need to use that version for now to get support with that controller. #966 Give that a shot as well.

QiuleiWang, 2019-12-19:
kextstat | grep 360Controller
  156    0 0xffffff7f84d48000 0xc000     0xc000     com.mice.driver.Xbox360Controller (1.0) 21410375-55C0-3491-8E39-173D2E8E4FF0 <58 27 6 5 3 1>

QiuleiWang, 2019-12-19:
sudo kextutil /Library/Extensions/360Controller.kext
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

Disabling KextAudit: SIP is off
(kernel) Notice - new kext com.apple.driver.usb.AppleUSBHostPlatformProperties, v1.2 matches prelinked kext but can't determine if executables are the same (no UUIDs).

QiuleiWang, 2019-12-19:
> From a fresh reboot, before connecting ANY devices to the system, can you run that command and see if you get the same result? Best guess is that you've somehow paired to a device, and your attempt [...]

kextstat | grep 360Controller
156 0 0xffffff7f84d48000 0xc000 0xc000 com.mice.driver.Xbox360Controller (1.0) 21410375-55C0-3491-8E39-173D2E8E4FF0 <58 27 6 5 3 1>

FranticRain, 2019-12-19:
@QiuleiWang Make a new issue, your problem is unrelated to the original post.

electronstudio, 2019-12-21:
Driver from the other thread works, thanks!

The output from command is the same though, with just keyboard and trackball connected:

```
> kextstat | grep 360Controller
> sudo kextutil /Library/Extensions/360Controller.kext
Password:
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Can't open CFBundle for /System/Library/Extensions/roller.kext.
Can't create kext from cache: /System/Library/Extensions/roller.kext is already open and has a different CFBundleIdentifier from identifier->path cache entry 162.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

Can't create kext from cache: /System/Library/Extensions/roller.kext is already open and has a different CFBundleIdentifier from identifier->path cache entry 162.
Disabling KextAudit: SIP is off
> kextstat | grep 360Controller
  155    0 0xffffff7f83498000 0xc000     0xc000     com.mice.driver.Xbox360Controller (1.0) 9868FCE0-505A-3056-8672-9ABEBCDF8BE5 <50 23 6 5 3 1>
```

closed 2019-12-21 by electronstudio

FranticRain, 2019-12-21:
Just to be sure, from which thread did you get the version that fixed your problem?

electronstudio, 2019-12-21:
https://github.com/360Controller/360Controller/issues/966#issuecomment-553721955