360Controller/360Controller #986: Wireless 360 controller not found

alimbada, 2019-12-11:
### Type of Controller
Xbox 360 Controller for Windows

### OS Version
10.13.6

### Driver Version
 1.0.0-alpha.3 and 1.0.0-alpha.5

### Connection Method
Wireless Adapter (using official wireless dongle)

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID: 0x0719
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller
Guide button keeps flashing (looking for adapter to pair with). Adapter doesn't switch on automatically; holding the button on the adapter switches it on at which point the light on the adapter flashes for a while then goes off.

### Response in Preference Pane
Preferences pane reports 'No device found'

### Detailed Explaination of Issue
Controller is not found. I have checked that the driver is enabled by ticking the box in the preferences pane.

cystbear, 2020-01-14:
Take a look https://github.com/360Controller/360Controller#wireless-xbox-360-controllers

```
As of macOS 10.11, Wireless Xbox 360 controller support causes kernel panics. This issue cannot be resolved with minor changes to the driver, and requires that the driver be re-written from scratch to resolve the issue. Due to an excess of caution, we have disabled Wireless Xbox 360 controller support as of 0.16.6. If you want to use a wireless controller, download 0.16.5 or earlier and disable the driver before the computer enters a "sleep" state in order to prevent kernel panics. Alternatively, you can revert to a macOS version before 10.11 to avoid this issue.
```

use `0.16.4` version to handle this issue https://github.com/360Controller/360Controller/releases/tag/v0.16.4
works fine for me

mightyx, 2020-01-28:
I use MacOS 10.14.6 and with the latest release the wireless Xbox 360 gamepad works perfectly with the original usb dongle, which I use through an usb to usb-c adapter. Thanks for your great work.

moubctez, 2020-03-06:
`WirelessGamingReceiver.kext` requires **com.apple.kpi.iokit**, **com.apple.kpi.libkern**, and **com.apple.kpi.mach** version 18.5, while 10.13.6 has version 17.7. Manually changing `Info.plist` lets the driver to load, but the controller is not visible in System Preferences.

proxygear, 2020-03-20:
Same problem :
* Controller : Wireless Adapter (using official wireless dongle)
* OS : 10.15.3
* Driver alpha-5
* Preferences pane reports 'No device found'

I tried to explore the console log, but as I'm not really sure what to look at ...