360Controller/360Controller #989: Can't enable driver, option to allow in "Security & Privacy" never appears

maxrothman, 2019-12-15:
### Type of Controller
Xbox 360

### OS Version
10.14.6, Macbook Pro (Retina, 15", mid-2015)

### Driver Version
1.0.0-alpha.5

### Connection Method
Wired

### Device Name and Info
>Device Name: Controller
<br/>Product ID: 0x028e
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller
Green ring lights up once then goes dark

### Response in Preference Pane
Device list says "No devices found"

### Detailed Explaination of Issue
I am unable to activate the driver. Here are the steps I took:
* Disable any potentially conflicting software, including Karabiner, BetterTouchTool, Alfred, etc.
* Reinstall driver (and restart the computer)
* Plug in controller
* Check "Enable driver" in the pref pane
* Click button switching view to "Security & Privacy"

Result: the button allowing me to allow the developer "Drew Mills" does not appear

Here's the output of `sudo kextutil /Library/Extensions/360Controller.kext`:
```
Error copying bundle during staging: -1
Unable to stage kext (/System/Library/Extensions/roller.kext) to secure location.
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Kext rejected due to system policy: <OSKext 0x7fd40dd98500 [0x7fffab9238e0]> { URL = "file:///Library/StagedExtensions/Library/Extensions/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
Kext rejected due to system policy: <OSKext 0x7fd40dd98500 [0x7fffab9238e0]> { URL = "file:///Library/StagedExtensions/Library/Extensions/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
Diagnostics for /Library/Extensions/360Controller.kext:
Warnings:
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done):
        ChatPadKeyboardEvents
```
Is there some way I can use `kextutil` to accept the developer certificate, or are there logs I can view to find out why the button doesn't appear?

FranticRain, 2019-12-15:
Every time you run `kextuil`, it should make the prompt re-appear within the "Security & Privacy" section of the preference pane. It will be at the bottom of the "General" tab. Give it a shot and let me know if it works for you.

maxrothman, 2019-12-15:
Nope, that had no effect.

FranticRain, 2019-12-15:
I just want to make sure that the kext is signed correctly. It should be, but if that prompt isn't showing up, I'm inclined to believe that this is a notarization error. Run this command:
```
spctl -a -v --type install /Library/Extensions/360Controller.kext 
```
It should give you this:
```
/Library/Extensions/360Controller.kext: accepted
source=Notarized Developer ID
```

Also if you're using alpha 5, then you should be able to check the "enable driver" checkbox in the "Advanced" settings of the driver. This will run a diagnostic that will probably tell you that you need to accept the signing certificate, but I want to make sure. It gives a slightly more granular error than `kextutil` does. So check that and let me know what it says as well.

maxrothman, 2019-12-15:
```shell
$ spctl -a -v --type install /Library/Extensions/360Controller.kext
/Library/Extensions/360Controller.kext: accepted
source=Notarized Developer ID
```
I am using alpha 5. As I described in my first message, when I originally checked the "enable driver" checkbox, it prompted me to switch to "Security & Privacy" and accept the certificate, where no accept button appeared. Now though, when I check the "enable" box System Preferences becomes unresponsive and the mouse switches to the beach ball of doom when it's over the app until I force-quit it.

FranticRain, 2019-12-15:
This sounds like it may be an issue with your mac, specifically. We can try and make sure that the driver/preference pane has no bearing on this. Go ahead and go through the process where it jumps you to the security pane. Then force quit out of system preferences, open it back up to that pane and click the button. It *should* still be there. if it still crashes, then your Mac is probably to blame in some way.

Of note is that devices and software that mess with inputs can break this button. Notably things like Wacom tablets and Karabiner Elements are known troublemakers. If you have any software or devices that do this sort of input editing, you might want to disable them, or unplug them, to see if they are causing this issue.

Moorviper, 2019-12-16:
click enable the driver and it should work.
Then at security allow the two extensions / the message that they have been blocked is there.
I have the same Macbook and 10.14.6 installed
at least it does for me.

[image]
[image]

gregrahg, 2020-03-17:
I am having the exact same issue as described the original post. Was this issue ever resolved?

clivend, 2020-03-20:
I am using alpha 5 on macos 10.15.3, when I click Enable Driver I get
[image]
then I open security preferences but all I see is
[image]
running `sudo kextutil /Library/Extensions/360Controller.kext`
gives
```
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Kext rejected due to system policy: <OSKext 0x7ffc59e0acc0 [0x7fff95100690]> { URL = "file:///Library/StagedExtensions/Library/Extensions/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
Kext rejected due to system policy: <OSKext 0x7ffc59e0acc0 [0x7fff95100690]> { URL = "file:///Library/StagedExtensions/Library/Extensions/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
Diagnostics for /Library/Extensions/360Controller.kext:
Warnings:
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done):
        ChatPadKeyboardEvents
```

would be really cool to have it working during this strange times locked at home, thank you!

FranticRain, 2020-03-20:
@clivend Do you have any software that modifies your inputs installed? Something like Karabiner? Or do you have a Wacom tablet installed? These types of devices can interfere with macOS’s willingness to let you approve security stuff.

clivend, 2020-03-23:
@FranticRain thank you for your reply! I tried to disconnect everything but had no luck :(

here is a list of my software currently installed in case that helps. nothing really modifies input to my knowledge

```[
    {
        "name": "Calendar",
        "version": "11.0"
    },
    {
        "name": "System Preferences",
        "version": "14.0"
    },
    {
        "name": "Preview",
        "version": "11.0"
    },
    {
        "name": "Activity Monitor",
        "version": "10.14"
    },
    {
        "name": "Installer",
        "version": "6.2.0"
    },
    {
        "name": "DiskImageMounter",
        "version": "559.40.3"
    },
    {
        "name": "App Store",
        "version": "3.0"
    },
    {
        "name": "ColorSync Utility",
        "version": "4.14.0"
    },
    {
        "name": "System Information",
        "version": "10.14"
    },
    {
        "name": "Calculator",
        "version": "10.14"
    },
    {
        "name": "AirPort Utility",
        "version": "6.3.9"
    },
    {
        "name": "AirDrop",
        "version": "1.0"
    },
    {
        "name": "Audio MIDI Setup",
        "version": "3.5"
    },
    {
        "name": "Archive Utility",
        "version": "10.15"
    },
    {
        "name": "Stocks",
        "version": "2.3"
    },
    {
        "name": "Photos",
        "version": "5.0"
    },
    {
[... 1362 lines cut ...]
    {
        "name": "Python",
        "version": "3.7.3"
    },
    {
        "name": "Java Remove",
        "version": "Java Remove"
    },
    {
        "name": "Python",
        "version": "2.7.15"
    },
    {
        "name": "countinstalltion"
    },
    {
        "name": "Creative Cloud Installer",
        "version": "4.3.0.256"
    },
    {
        "name": "Java Mission Control",
        "version": "6.0.0"
    }
]
```

somm-doe, 2020-05-11:
I am having the same issue - with other apps aswell though.

pwstaten, 2020-09-20:
Also unable to enable driver. When I installed 360controller the first time, it worked (wired controller). A week or so later, the controller stopped working. In the preferences pane, I could not check "Enable Driver", without getting sent to the security and privacy pane. But the option to enable Drew Mills never showed up. (And I swear I already enabled Drew Mills when I first installed the software.) If I enable apps from anywhere (temporarily), checking the box still doesn't work, so I get the feeling that checking the box isn't even running (or attempting to run) any binary files at all.

pwstaten, 2020-09-21:
Also, I ran spctl to check the notarization:

spctl -a -v --type install /Library/Extensions/360Controller.kext 
/Library/Extensions/360Controller.kext: accepted
source=Notarized Developer ID

And I tried force-quitting out of the preferences pain as FranticRain suggested, to no avail. So if something is wrong with maxrothman's mac, then something similar must be wrong with mine. (And this is with no USB peripherals.)

I suspect a system policy is preventing the kext from loading.

maxrothman, 2020-11-20:
For what it's worth, I got the "enable driver" box to stay checked by following [these instructions](https://github.com/pqrs-org/Karabiner-Elements/issues/1785#issuecomment-484591680). The team ID for 360controller is G947KKM5RL.

maxrothman, 2020-12-27:
I just installed [1.0.0-alpha6](https://github.com/360Controller/360Controller/releases/tag/v1.0.0-alpha.6) and now my controllers work fine. I don't know exactly what changed, but I'm glad it fixed my issue!

closed 2020-12-27 by maxrothman