360Controller/360Controller #990: Rock Candy Controller

cancan-boop, 2019-12-18:
### Type of Controller
Xbox 360

### OS Version
10.15.1

### Driver Version
Version 1.0.0-alpha.5

### Connection Method
Wired

### Device Name and Info
>Device Name: Rock Candy
<br/>Product ID: 0x011f
<br/>Vendor ID: 0x0e6f

### Response of Controller
The guide button lights up once and does not light up again unless taken out and plugged back in. In other driver versions I've tested (alpha 2 and possibly 3), it continues to blink when plugged in.

### Response in Preference Pane
The controller does not appear at all. Simply says "No devices found"

### Detailed Explaination of Issue
I'm not sure how much more info to give, especially with the limited feedback the controller gives. I checked and unchecked the "Enable Driver" button, but it does not seem to make any difference. I hvae tested all alpha driver versions and cannot get the controller to show up. I have done it before a year ago, but I don't know why it isn't working anymore. I've uninstalled and installed each driver multiple times without much change. Please help. Thanks

FranticRain, 2019-12-18:
So this device is definitely in the plist, so it should be recognized by the driver. The fact that the preference pane never recognizes it is confusing. Just so I can be 100% sure that the driver is behaving as expected, can you install the latest version of the driver and then run this command in your terminal so I can check if the driver is being loaded correctly? It will ask for your logon password, but that's just because it's a macOS utility that needs administrator privileges.
```bash
sudo kextutil /Library/Extensions/360Controller.kext
```

cancan-boop, 2019-12-19:
So I uninstalled everything, restarted the computer, and reinstalled the alpha 5 driver, and restarted once more. I then pasted that text into the terminal and here is everything that proceeded:

Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

(kernel) Notice - new kext com.apple.driver.usb.AppleUSBHostPlatformProperties, v1.2 matches prelinked kext but can't determine if executables are the same (no UUIDs).