360Controller/360Controller #991: NVIDIA GeForce Now

TheBeverage, 2019-12-20:
### Type of Controller
Wireless 360 Controller

### OS Version
macOS Mojave 10.14.6

### Driver Version
Replate this text with the version or versions of the driver that you are using or have tested

### Connection Method
Official Microsoft Wireless Adapter for PC

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID:
<br/>Vendor ID:

### Response of Controller
None.

### Response in Preference Pane
Nothing unusual. Changing settings does not affect the issue unless the controller is disconnected entirely, in which case the issue does not exist.

### Detailed Explanation of Issue
When trying to launch a game via NVIDIA GeForce Now with a wireless 360 controller connected via this driver, any attempt to launch crashes the game. I have no such issues with using the controller in local Steam games, nor with any other controllers. I don't know if this is fixable on your end, but if not, I am wondering if perhaps the controller spoofing could be modified in such a way that it reads as a controller supported for GFN on Mac, such as the XB1 pad?

I have attached the full error in a text file.
[gfn-error.txt](https://github.com/360Controller/360Controller/files/3989673/gfn-error.txt)

FranticRain, 2019-12-20:
The driver does not provide any sort of compatibility for individual games beyond what is provided in the preference pane. The driver simply sends button data to the operating system. Once it does that, control is out of the driver's hands. Since Apple does not provide any standardized means of interacting with controllers in macOS, every game uses a different method to read controller data. We cannot provide individual fixes for each game within the controller. That simply isn't feasible. It is the responsibility of game developers to support controllers, not the other way around. We cannot and will not provide support for individual games. If you are using an Xbox One controller, we can recommend that you use the "Pretend to be an Xbox 360 Controller" option in the "Advanced" menu for better game compatibility. Beyond that, there is nothing we can do.

FranticRain, 2019-12-20:
But, if you were curious, this error dump is telling you that whatever the heck this thing is, is the problem:
```
Thread 31 Crashed:: WINMM gamepad monitor
0   libGeronimo.dylib             	0x000000010c3a6623 XBOX_destroyWirelessDevice + 339
1   libGeronimo.dylib             	0x000000010c3a5238 XBOX_addDevices + 808
2   libGeronimo.dylib             	0x000000010c3a4ee6 XBOXDeviceInit + 150
3   libGeronimo.dylib             	0x000000010c3a034f DEV_deviceMonitorThread + 191
4   libGeronimo.dylib             	0x000000010c3a418b WINHELP_threadWrapper + 11
5   libsystem_pthread.dylib       	0x00007fff698812eb _pthread_body + 126
6   libsystem_pthread.dylib       	0x00007fff69884249 _pthread_start + 66
7   libsystem_pthread.dylib       	0x00007fff6988040d thread_start + 13
```

closed 2019-12-20 by FranticRain