360Controller/360Controller #992: Unknown error -603946986 when attempting to enable driver

amizener64, 2019-12-21:
### Type of Controller
Xbox One

### OS Version
MacOS Sierra 10.12.6

### Driver Version
1.0.0-alpha.5

### Connection Method
Standard USB

### Device Name and Info
Xbox ONE Pro Ex controller:

  Product ID:	0x543a
  Vendor ID:	0x24c6
  Version:	1.01
  Serial Number:	0000458B870F6900
  Speed:	Up to 12 Mb/sec
  Manufacturer:	BDA
  Location ID:	0x14100000 / 2
  Current Available (mA):	500
  Extra Operating Current (mA):	0

### Response of Controller
No response; it's as if I didn't plug it in.

### Response in Preference Pane
None. Not detected by driver.

### Detailed Explaination of Issue
Hello! The installation of the driver seemed to go smoothly. However, although my Mac recognizes that my controller has been plugged in, the driver does not. It tells me "No devices found", and the "Enable Driver" box is unchecked by default under the 'Advanced' pane. When attempting to check the 'Enable Driver' box, I encounter the following message:
/Users/andrewmizener/Desktop/Screen Shot 2019-12-20 at 11.37.08 PM.png

amizener64, 2019-12-21:
Since the image didn't seem to attach, I'll paste the contents here:
"The driver has failed to load due to an unknown error, code: -603946986. Please report this error to the driver maintainer."

FranticRain, 2019-12-22:
This version of the driver should solve your problem.

[360ControllerInstall.dmg.zip](https://github.com/360Controller/360Controller/files/3993107/360ControllerInstall.dmg.zip)

closed 2019-12-22 by FranticRain