360Controller/360Controller #993: Xbox One Controller detected but no input detected after firmware update

mchlbowyer, 2019-12-23:
### Type of Controller
Xbox One

### OS Version
macOS 10.15

### Driver Version
0.16.11 and 1.0.0 alpha 5

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID: 0x02ea
<br/>Vendor ID: 0x045e

### Response of Controller
Guide button lights up (solid light), vibrates when connected

### Response in Preference Pane
Shows connected controller but no input detected

### Detailed Explaination of Issue
The controller was working fine, however then I updated the firmware for the controller (as I was having issues with bluetooth connection and thought this may resolve it) - since the firmware update no input is detected in games on macOS and no input is shown on Controller test. I have also tested using the 'Pretend to be an xbox 360 controller' and still no input is detected. Thanks in advance

FranticRain, 2019-12-23:
This could be a pretty complicated issue. I'll need to see if I can reproduce it when I get a chance. In the meantime, we can check some simple stuff just to make sure. Are you connected via both a wire and Bluetooth at the same time? That might be causing issues with the driver.

exgphe, 2019-12-28:
Same issue here with the same product id and vendor id. After I updated the controller firmware to 4.8.1923.0 via a Windows machine, it shows up in the preference pane but no input is detected. I have disabled the bluetooth of macOS and same problem.

L0VEFIRE, 2019-12-28:
I did some research on this, since I'm now very cautious to update my firmware. I still have Xbox Controller (1708) _Firmware 3.1.1221.0_, and my controller works playing games if the controller is ON before launching them.

**> FranticRain (This might help you)**
I found a link on the patch notes for the firmware "4.8.1923.0" that is causing everyone issues
https://answers.microsoft.com/en-us/xbox/forum/all/new-release-of-controller-update-23-october-2019/4ce5abe6-86f8-4490-bda6-c3943c31713e?auth=1

FranticRain, 2019-12-28:
I still haven’t had a chance to look at this one, but that change log is just Bluetooth changes. I’ll need to do a full packet sniff to see what the story is.

mchlbowyer, 2019-12-28:
@FranticRain sorry I did respond on my phone but just realised it didn't go through. No only connected via cable, had bluetooth turned off on laptop.

After a bit more investigation though I have found a bit more info (and this may actually just be my controller but not sure) - it only seems to happen after I have had it connected to something via Bluetooth - if I connect via bluetooth and then disconnect/turn off controller, when it is plugged in via cable again it won't pick it up. To fix this, I have to connect the controller to a windows machine, tell it to check for a firmware update, then once that's done I can connect back to the Mac and it will pick up the inputs. A very long process to get it working again, but I only have one controller so cannot test if its the controller or something else.

FranticRain, 2019-12-28:
Having a reproducible set of conditions should help a lot, thank you!

FranticRain, 2020-01-02:
Did some cursory work on this and was able to get the steps to reproduce down to this:

1. Connect controller via Bluetooth to something (in my case, a Windows PC)
2. Connect the controller via USB to a Mac. Controller will appear in the preference pane but not respond.
3. Disconnect USB from Mac, connect to Windows PC, allowing to pair and finish setup on Windows.
4. Connect USB to Mac, input is now available.

It looks like there's just an initialization packet that I am missing that will let the driver take over the Bluetooth connection and work as expected. In my current capacity, I can't grab it yet. But when I get a chance back at my normal system, this should be an easy fix. Just wanted to keep you all updated.

mchlbowyer, 2020-01-06:
@FranticRain thanks for the update :)

closed 2023-02-22 by mchlbowyer