360Controller/360Controller #994: @Play 360 controller not detected

ghoster2020, 2019-12-25:
### Type of Controller
@Play 360 wired controller

### OS Version
10.14.6

### Driver Version
Version 1.0.0-alpha.5

### Connection Method
Wired

### Device Name and Info
Product ID:	0x0006
Vendor ID:	0x0079
 Version:	1.09

### Response of Controller
Detected as 'generic joystick' and works in steam 'big picture' overlay.

### Response in Preference Pane
Not detected.

### Detailed Explaination of Issue
Not detected.
sudo kextutil /Library/Extensions/360Controller.kext
Error copying bundle during staging: -1
Unable to stage kext (/System/Library/Extensions/roller.kext) to secure location.
kext file:///Library/StagedExtensions/System/Library/Extensions/B0DCE874-42F9-4BEE-8945-8AF096557D4F.kext/ is in hash exception list, allowing to load
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

FranticRain, 2019-12-25:
This driver version should work for you. Let me know if it doesn't.

[360ControllerInstall.dmg.zip](https://github.com/360Controller/360Controller/files/4000820/360ControllerInstall.dmg.zip)

ghoster2020, 2019-12-28:
Still not detected.

FranticRain, 2019-12-28:
Okay. Then this is what we call a “counterfeit” controller. Some Chinese controllers are made to look like Xbox controllers, but don’t work with the consoles. We don’t support these kinds of controllers, sorry.

closed 2019-12-28 by FranticRain