360Controller/360Controller #996: Powering Controller after games are launched results in (No Input & Crashes Game)

L0VEFIRE, 2019-12-28:
### Type of Controller
Xbox One (1708)

### OS Version
macOS Catalina 10.15.2

### Driver Version
Version 1.0.0-alpha.5

### Connection Method
Wired (USB-C to Micro USB)

### Device Name and Info
**USB 3.1 Bus:**

  Host Controller Driver:	AppleIntelCNLUSBXHCI
  PCI Device ID:	0xa36d 
  PCI Revision ID:	0x0010 
  PCI Vendor ID:	0x8086 

**Controller:**

  Product ID:	0x02ea
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	3.01
  Serial Number:	3033363030353037323632383239
  Speed:	Up to 12 Mb/s
  Manufacturer:	Microsoft
  Location ID:	0x14100000 / 6
  Current Available (mA):	500
  Current Required (mA):	500
  Extra Operating Current (mA):	0

### Response of Controller
When the controller is On before a game is launched it works. BUT ... When a game is launched before the controller is powered ON, then there is no controller input and when I quit my games they will crash not only the game but also steam®, I tried this will my entire library and all my games do this. so curious if this driver is the culprit since I never used the alpha drivers before I have always used version 0.16.10. (I haven't tested the difference yet)

### Response in Preference Pane
Loads in Dark mode for a second, then switches back to white background. (reported in different thread)

### Detailed Explaination of Issue
When I launch steam® and plug in (or Power On) my controller and then start a game it works fine.
When I launch steam® and then launch a game also, and then plug in (or Power On) my controller there will be no controller input with the game, (some games it will causes the games buttons to act erratic (clicking settings will maybe not click, or will open something else, or crash the game), overall quitting the game will crash both the game and the Steam® app.

not sure if this is the driver or steam® or what but still needs to be reported.

FranticRain, 2019-12-28:
This is a game specific issue. There is nothing we can do in the driver that will fix this. Speak to the game developers if you want a solution.

closed 2019-12-28 by FranticRain

L0VEFIRE, 2020-01-02:
Hi FranticRain, 

you said that is had to do with the developers. so I uninstalled the driver, then tested the same method that was causing steam (standalone) and also when in games to crash. suffice to say, neither crashed once after x25 tries. So can you explain to me why having you drive installed is the only time it causes my Steam app to crash as well as my games when turning it on after they loaded, or unplugging it after a games as been loaded. 

im not sure that can be pin 100% on the dev's but I'm just asking.