360Controller/360Controller #997: Uninstalling Driver gives ERROR

L0VEFIRE, 2019-12-28:
##Type of Controller

Xbox One (1708)

##OS Version

macOS Catalina 10.15.2

##Driver Version

Version 1.0.0-alpha.5

##Connection Method

Wired (USB-C to Micro USB)

##Device Name and Info

**USB 3.1 Bus:**

Host Controller Driver:	AppleIntelCNLUSBXHCI
PCI Device ID:	0xa36d
PCI Revision ID:	0x0010
PCI Vendor ID:	0x8086

**Controller:**

Product ID:	0x02ea
Vendor ID:	0x045e (Microsoft Corporation)
Version:	3.01
Serial Number:	3033363030353037323632383239
Speed:	Up to 12 Mb/s
Manufacturer:	Microsoft
Location ID:	0x14100000 / 6
Current Available (mA):	500
Current Required (mA):	500
Extra Operating Current (mA):	0

### Response of Controller
N/A

### Response in Preference Pane
N/A - see below

### Detailed Explaination of Issue
I open the xbox pref. pane and clicked uninstall, I got a prompt, then click okay, then it gave me an error. "There was an error in the 360 Preferences". Then when I clicked on the xbox pane it said it didn't exist, so I quit sys. preferences and re-opened it, then it was gone.  but I'm not sure how gone it really is, do to the error and why I'm getting this every time I uninstall.
[Uninstaling xbox driver.zip](https://github.com/360Controller/360Controller/files/4007161/Uninstaling.xbox.driver.zip)

FranticRain, 2019-12-28:
Without any specific error code, I can’t be sure why the uninstall failed. But if you go to the releases page, the Alpha 5 release includes an “uninstall.command.” Download and run that. It is a complete uninstaller that should clean up any remaining pieces.

closed 2019-12-28 by FranticRain

L0VEFIRE, 2019-12-28:
That shell command doesn’t work on catalina 10.15.2 says command doesn’t have privlages

FranticRain, 2019-12-28:
Open up your terminal and run it with `sudo ./uninstall.command`

It should have prompted you for a password when you double clicked on it. If it doesn’t have permissions, then you’re not an admin on your own machine.

L0VEFIRE, 2019-12-29:
This is the message I get. And I am admin.