360Controller/360Controller #998: CONFIRMED: Driver is causing macOS to hang during reboot/bootup + degraded system performance.

L0VEFIRE, 2020-01-02:
### Type of Controller
Xbox One (1708)

### OS Version
macOS Catalina 10.15.2

### Driver Version
Version 1.0.0-alpha.5

### Connection Method
Wired (USB-C to Micro USB)

### Device Name and Info
**USB 3.1 Bus:**

  Host Controller Driver:	AppleIntelCNLUSBXHCI
  PCI Device ID:	0xa36d 
  PCI Revision ID:	0x0010 
  PCI Vendor ID:	0x8086 

**Controller:**

  Product ID:	0x02ea
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	3.01
  Serial Number:	3033363030353037323632383239
  Speed:	Up to 12 Mb/s
  Manufacturer:	Microsoft
  Location ID:	0x14100000 / 6
  Current Available (mA):	500
  Current Required (mA):	500
  Extra Operating Current (mA):	0

### Response of Controller
(side issue) Only works in pane, crashes Steam® app when power after steam is open, no input after games are launched, or input is randomized before or after game is launched. unplugging controller when the game is open or steam® app (standalone) is open crashes them. (this doesn't happen when driver is not installed) 

### Response in Preference Pane
controller works

### Detailed Explaination of Issue
(side note - Explanation is spelled wrong above)

MAIN ISSUE:
anyway, more to the point here. I have tested this pretty throughly, and after x15 reboots without the driver installed, and using the macOS. The Mac boot down process was flawless (the dock smoothly side off screen, menu bar does the same, and the desktop smoothly transitions' to a black screen. then reboots. when using the macOS with apps they load faster and are more responsive.

The moment I installed the driver, the boot down process (restart), the dock hanged, the desktop transition didn't happen, just desktop and black screen, using apps were delayed launching, and hanged more. I rebooted the Mac 5 more times the same thing happened. 

then uninstalled it again, and rebooted, everything was back to normal (see first paragraph).

all this time I was point the finger at apple, and I find out that its the driver not working causing chaos on my machine. it even doesnt work anymore with my steam games. I say some investigating Is in order. since this doesnt happen when I'm not using the driver.

FranticRain, 2020-01-02:
I'm going to respond to a couple issues since you seem to really like to shotgun your issues out there, so I'm going to try and focus in here since you've mentioned both.

I'll start with the question you posed in #996:

>Hi FranticRain,
>
>you said that is had to do with the developers. so I uninstalled the driver, then tested the same method that was causing steam (standalone) and games to crash. Suffice to say, neither crashed once after x25 tries. So can you explain to me why having you driver installed is the only time it causes my Steam app to crash as well as my games when turning the controller on after they loaded, or unplugging it after a games as been loaded.
>
>im not sure that can be pin 100% on the dev's but I'm just asking.

This is how I see it happening. When the controller is connected to the system via the driver, it tells the OS that a new controller has been added. Then the OS can advertise the controller. That's all the driver does. It just augments the OS. If the OS crashes, then it's the driver's fault. It has no bearing on any single app on your computer.

If the application crashes only when the driver is connected after the application is started, then the application is failing to process the operating system adding a new device. Once again, if there was a bug in the driver, it would cause the system to kernel panic. You'd have to reboot. 

> ...crashes Steam® app when power after steam is open...

I just tested this and I cannot reproduce the error.

>...no input after games are launched, or input is randomized before or after game is launched...

Game-specific issue. Once again we only provided a connection to the OS. Games not working is not something we can fix.

>unplugging controller when the game is open or steam® app (standalone) is open crashes them.

This is a game experiencing a sudden loss of an expected interface from the OS. It isn't surprising that this would crash a game. Once again, a driver bug would cause an OS crash. An application bug causes an application crash.

>(this doesn't happen when driver is not installed)

Of course it doesn't, because the controller isn't added to the OS, the game never sees it, and therefore doesn't care when the controller is disconnected. This is not indicative of a bug in the driver at all. I'll say it again: Driver bugs cause OS crashes.

>The moment I installed the driver, the boot down process (restart), the dock hanged, the desktop transition didn't happen, just desktop and black screen, using apps were delayed launching, and hanged more. I rebooted the Mac 5 more times the same thing happened.

I haven't noticed this at all. This may be an issue with your machine specifically. I will continue testing to determine if I can confirm your issue.

That brings me to another point. The "CONFIRMED" in your title. Whether or not something is "confirmed" is a term we use in our issues when an issue can be confirmed across multiple systems, implying that it is not an individual issue and is therefore a potential issue with the driver. As such, I will be removing the "CONFIRMED" from your title, because I have never heard of anyone else complaining about the issues. Furthermore, I have also attempted to replicate your issues and I am unable to do so, further disqualified your "CONFIRMED."

This issue will be marked as "needing confirmation," which means that we need to see if anyone else is experiencing similar issues, so we can try and figure out what the common link is that is causing your issue. In your case, it is likely another bad actor on your machine that is being aggravated by the installation of the driver. This misbehaving app just doesn't misbehave if the driver isn't present, for some reason. Or it could be that the driver really is the problem and that the app just misbehaves because the driver is messed up. Either way, we need to figure out what the misbehaving app is, and we can only do that if we can find another user with your same problem.

L0VEFIRE, 2020-01-03:
First I wanted to apologize for the impulsive title "confirmed", which was not accurate on my part, I have been going crazy trying to solve some issue on my new Mac and I just felt short of pure coincident that it only seemed to be happened after I installed that driver, but that didn't mean it was the driver. 
So I loaded up my test partition account, fresh, and reboot a couple time, low and behold the same problem happen almost every time. so defiantly the macOS for this machine specifically and not your driver from what I am seeing. You know you stuff and sorry my confusion got laid onto your hard work. 

I wanted to clear this up. take care!

closed 2020-01-03 by L0VEFIRE

FranticRain, 2020-01-03:
Glad to hear you figured it out. Happy to help out.