360Controller/360Controller #1003: Xbox One controller "no devices found"

JunjieHu, 2020-01-14:
### Type of Controller
Xbox One Wireless Controller from Microsoft

### OS Version
macOS Mojave 10.14.6

### Driver Version
I tried both 0.16.5, 0.16.11 and the lastest 1.0.0-alpha.5. None of them work.

### Connection Method
I tried both Wired connection by USB, and Wireless connection by bluetooth.

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Bluetooth USB Host Controller:
<br/>Product ID:	0x828c
<br/>Vendor ID:	0x05ac (Apple Inc.)

### Response of Controller
The guide button lights up. It goes vibrate when I connect to the usb cable. 

### Response in Preference Pane
"No devices found"

### Detailed Explaination of Issue
I tried the following commands, but it didn't work.
$ sudo kextcache --clear-staging

$ sudo kextutil /Library/Extensions/360Controller.kext
Error copying bundle during staging: -1
Unable to stage kext (/System/Library/Extensions/roller.kext) to secure location.
kext file:///Library/StagedExtensions/System/Library/Extensions/EAFA3039-2C53-4788-B931-08D16329A74E.kext/ is in hash exception list, allowing to load
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings:
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done):
        ChatPadKeyboardEvents

I also make sure the Security & Privacy section "Allows apps downloaded from anywhere" by typing this in the terminal:
$ sudo spctl --master-disable

I also turn off the Firewire in the Security & Privacy section. But nothing works. Could you help? Thanks!

Retch, 2020-02-06:
I have the same problem.

iwan41, 2020-03-06:
I have the same problem.
In my case, I was playing a game and the controller stopped working. After that system preferences gives "no device found"

standbylights, 2020-03-27:
i'm having the same issue, it connects via bluetooth and when i plug it in it vibrates, but it still says no devices found. i can use my controller in steam in big picture mode, but not in any games.