360Controller/360Controller #1004: Amazon Basics Xbox One Pro Ex wired controller does not appear in Preferences Pane

zdinardo, 2020-01-15:
### Type of Controller
Xbox One

### OS Version
macOS 10.15.2

### Driver Version
0.9.1 
1.0.0-alpha_5 
0.15 (beta 5) (from the google drive link in #701) 

### Connection Method
Wired

### Device Name and Info
Xbox ONE Pro Ex controller:

Product ID: 0x543a
Vendor ID: 0x24c6
Version: 1.01
Serial Number: 00002BCFE72AD10E
Speed: Up to 12 Mb/sec
Manufacturer: BDA
Location ID: 0x14400000 / 3
Current Available (mA): 500
Current Necessary (mA): 500
Extra Operating Current (mA): 0

### Response of Controller
The small light near the Xbox button does not light up. 

### Response in Preference Pane
The controller does not appear in the preference pane. "No devices found" is the only scroll menu option.  

### Detailed Explanation of Issue
The first time I used this driver (with the same controller on a previous version of macOS, I think Mojave) I had the same problem. It was solved with the google drive link in #213, but since I upgraded OS (I think, this was a little while ago) that version doesn't work anymore. In #701, @FranticRain suggested running `sudo kextload file.kext` (replacing "file" with where the thing is" to see if there would be a Security & Privacy issue, but I don't receive any terminal output. Unfortunately I have neither another Xbox One controller nor an Xbox One to get a positive control on the hardware. Not sure what would have happened between macOS versions, but thanks in advance for any help you could offer!

FranticRain, 2020-01-15:
So this PID/VID is in the device list not once, but twice. So you definitely aren't having an issue with the driver not recognizing the controller. I want to make sure that the driver is installed correctly, so please run this command in your terminal and let me know what it says:
```
sudo kextutil /Library/Extensions/360Controller.kext
```

This issue could actually be caused by the OS if you were previously using a version of macOS before 10.11. Do you remember what version you were on before?

zdinardo, 2020-01-16:
I get this as the output: 
`Untrusted kexts are not allowed
Kext with invalid signature (-67062) denied: /Library/StagedExtensions/Library/Extensions/E6C5C652-FEE4-497A-88B3-2DC0F141C982.kext
Bundle (/Library/Extensions/SteamInputX.kext) failed to validate, deleting: /Library/StagedExtensions/Library/Extensions/E6C5C652-FEE4-497A-88B3-2DC0F141C982.kext
Unable to stage kext (/Library/Extensions/SteamInputX.kext) to secure location.
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

(kernel) Notice - new kext com.apple.driver.usb.AppleUSBHostPlatformProperties, v1.2 matches prelinked kext but can't determine if executables are the same (no UUIDs).`

Is the "different from containing kext's" thing something important? I've installed multiple versions and it could be reading different things from different places. In terms of the "Untrusted kexts are not allowed," I've definitely done the thing where I allow apps from identified developers (including you), so I'm not sure what that could be, either? 

My macOS version when I originally had it working was, to the best of my knowledge, High Sierra (10.13), judging by the release date of the OS and when I bought the controller. I usually keep my computer pretty up to date so I'd be surprised if I had been running something before 10.11 while 10.13 was released.

FranticRain, 2020-01-16:
So "differs from containing kext's" is normal. And the "Untrusted kexts are not allowed" message is from a completely different driver that depends on `360Controller.kext` (SteamInputX.kext).

It looks like this is a classic issue with the IOUSBHost upgrade. Apple changed their driver setup at some point and re-implemented how the drivers work under the covers. Part of this upgrade meant that PowerA and PDP devices stopped working. I don't know why, Apple doesn't know why. It's just how it is. But seeing as the code works for other devices, and all devices should work exactly the same, per the Xbox spec, I'm tempted to say that it's Apple's fault and there's nothing I can do about it. Sorry.

I'm going to leave this issue open for a bit since I was, at one point, working on changing the initialization packets up a bit to solve a problem. If there is anything the driver can do to solve the problem, it will be fixed as part of that effort. So I'm going to leave this open until I can get a build out to you that would potentially get this to work.

zdinardo, 2020-01-16:
Ok, thanks for all your help! I'll stay up to date on things and will post if it gets fixed.

Andrefei, 2020-07-02:
I am using the exact same controller (prod/vendor: 0x543a/0x24c6) with macOS 10.15.5, I had some problems with earlier releases, but the controller is working with version 1.0.0-alpha.6  🎉🎉

FranticRain, 2020-07-02:
Awesome. I guess macOS resolved their issue. Closing this out.

closed 2020-07-02 by FranticRain