360Controller/360Controller #1005: Driver Not Recognising Xbox 360 Controller

eawbell, 2020-01-16:
### Type of Controller
Xbox 360

### OS Version
Sierra 10.12.6 (mac)

### Driver Version
360Controller 1.0.0-alpha.5

### Connection Method
Wired (USB)

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Microsoft Xbox 360 Controller
<br/>Product ID: 0x028e
<br/>Vendor ID:0x045e  (Microsoft Corporation)

### Response of Controller
Green circle of light around the X in the middle of the controller once (immediately upon plugging in the controller) and then nothing. 

### Response in Preference Pane
No devices found

### Detailed Explanation of Issue
I plug in the controller, green light on controller comes on as stated above, and open the preferences pane but the controller isn't there. I've tried unplugging and plugging it back in again into different USB ports. Also tried clicking the 'enable driver' button in the advanced tab but that returned the following error: The driver has failed to load due to an unknown error, code: -603946986. Please report this error to the driver maintainer.

Fair warning - I'm not a coder or super computer whizz (yes, I'm aware i sound like a grandpa even though I'm not.) I can follow instructions though. Please help :)

FranticRain, 2020-01-16:
So this is a quick fix. Versions of macOS have a little issue with the latest version of the driver, and I haven't had the time to get all of the fixes I want to put in to the next patch rolled out yet. So here's a version of the driver that should work for you. I don't know if I've had anyone as far back as 10.12, but just install this, give it a shot, and let me know how it works out for you. [360ControllerInstall.dmg.zip](https://github.com/360Controller/360Controller/files/3993107/360ControllerInstall.dmg.zip)

eawbell, 2020-01-16:
Awesome! Thank you :) I'll give it a go now and keep you updated. Do I need to uninstall the other one first or can I just install the one you sent straight away?

eawbell, 2020-01-16:
I uninstalled then installed the one you sent and it worked beautifully. Thank you!

closed 2020-01-16 by eawbell