360Controller/360Controller #1008: Are TWO wired xbox 360 controllers supposed to work in parallel?

schlichtanders, 2020-02-08:
### Type of Controller
Xbox 360 afterglow

### OS Version
10.15.2

### Driver Version
360Controller 1.0.0-alpha.5

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Afterglow Gamepad for Xbox 360
<br/>Product ID: 0xfafd
<br/>Vendor ID: 0x24c6

### Response of Controller
sometimes the button does not blink, but this time both blinks

### Response in Preference Pane
all work as expected

### Detailed Explaination of Issue
While the driver finds both, It seems like only a single controler is recognized by a steam game (divinity2). I.e. it finds correctly one, but we cannot join with the other.
Is this intended bevhaiour that only one controller is supported? 
(I couldn't find any mention on how many controllers actually should work in the README, neither I found another issue about this, hence I opened this new issue)

FranticRain, 2020-02-08:
The driver supports usage of multiple controllers at the same time. This may be a game specific issue. The way you can make sure is to check in an HTML5-based gamepad tester like this one: https://html5gamepad.com/. If you can see both controllers in that tester, then the driver is working as expected and providing both controllers to the system. If you only see one, then there might be a bug in the driver. Let me know the results of your testing so I can appropriately categorize this issue.

schlichtanders, 2020-02-09:
thank you very much! I tested that web-page and indeed, like you said, both controllers are recognized and working as soon as I click on one of them.

I will contact game support then. Thanks.

closed 2020-02-09 by schlichtanders