360Controller/360Controller #1012: Unable to Uncheck "Enable Driver"

daaamon, 2020-02-12:
### Type of Controller
Xbox One

### OS Version
Mojave 10.14

### Driver Version
Version 1.0.0-alpha.5

### Connection Method
Wired

### Device Name and Info
N/A

### Response of Controller
Controller responds fine.

### Response in Preference Pane
Everything acts as expected

### Detailed Explaination of Issue
I am trying to uncheck the "Enable Driver" button under Xbox 360 Controllers>Advanced so that I can play a game on Steam (Geometry Wars 3: Dimensions). I read on the Steam forums that i must uncheck this box if i want the game to work with my xbox controller. When I check the box, however, I am getting an error that says "The kext cannot be unloaded because it is retained by macOS. You will need to reboot your computer before you can unload the kext."

After that, the xbox controller no longer becomes recognized under the "Xbox 360 Controllers" menu and the "Enable Driver" button remains checked. At this point, if i unplug and re-plug in the controller, I get the same result- it's not recognized by my computer at all.

The only way for the controller to become recognized again is if i reboot my computer.

P.S. There is a little note under the "Enable Driver" box that says "Due to some internal limitations, you need to connect your device once to be able to disable this." What does this mean exactly?

**Damon**

FranticRain, 2020-02-12:
So what that line is saying is basically shorthand for this:
In order to “disable” the driver, we have to forcibly unload it from macOS. We can’t unload the driver until it’s been loaded once, and every time you reboot, it will be loaded on the first connection of a controller. So that’s what it’s trying to tell you in so many words. Now the strange part here is that the symptoms you are giving are identical to what “disabling” the driver does. The driver ceases to manage the connection to the controller and the controller is now at the whims of a game to handle any sort of recognition of the controller.

What that warning is telling you is that the driver couldn’t be unloaded because something was using it at the time. The operating system tried to retain the connection. But if the OS retained the connection, it shouldn’t be acting like this. Ignore the warning and see if the controller still works with the game, because it seems like it’s working as intended.

daaamon, 2020-02-12:
> So what that line is saying is basically shorthand for this:
> In order to “disable” the driver, we have to forcibly unload it from macOS. We can’t unload the driver until it’s been loaded once, and [...]

First off, thank you for the reply. Second off, I thought that it also implied that the driver was "unloaded" after it failed to be recognized by my preferences... but it seems that my computer loses all connection to it at that point. It no longer lights up, or vibrates, or anything. Do you think that uninstalling the driver completely will maybe fix the issue?

**Damon**

FranticRain, 2020-02-12:
The driver is the part responsible for sending the data that starts the controller. This process lights up the logo, vibrates the controller, and performs all operations related to the control. Without the driver, the operating system will treat it just like a brick. An application on the machine can connect directly to the controller and manage this initialization and control itself. Games that do this is the reason for this "disable" checkbox.

andyfromminneapolis, 2020-02-21:
Hey there! 
Im using OS 10.13.2, High Sierra and when I try to click the box that says Enable Driver, I get this: "The driver has failed to load due to an unknown error, code: -603946986. Please report this error to the driver maintainer."
Any thoughts? Thanks!

FranticRain, 2020-02-21:
@andyfromminneapolis Download the driver from this issue. It will solve your problem. https://github.com/360Controller/360Controller/issues/1011