360Controller/360Controller #1013: Wired XBOX 360 Controller - No Devices Found - Catalina

Unboundclassic, 2020-02-15:
### Type of Controller
XBOX 360

### OS Version
macOS 10.15.3 (19D76)

### Driver Version
360Controller 1.0.0-alpha.5
360Controller 0.16.11 (preference pane didn't show up)
360Controller 0.16.11 Notarized (current install)

### Connection Method
Original Wired Microsoft controller. Also tried 3rd party knockoff, same result. They are connected via a USB to USB-C adaptor. I had previously used this setup in Mojave with no issues. 

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Controller 
<br/>Product ID: 0x028e
<br/>Vendor ID: 0x045e (Microsoft Corporation)

### Response of Controller
With 360Controller 0.16.11 the guide button flashes endlessly. With 360Controller 1.0.0-alpha.5 the guide button would flash briefly when plugged in. 

### Response in Preference Pane
"No devices found."

### Detailed Explanation of Issue
The controller will not show up in the preference pane. "No devices found." There are no warnings in the System Preferences security pane. I did allow the extension. I tried the "Enable Driver" option in the Advanced tab. No difference. And now it won't turn off permanently, lol. 

In another issue I saw someone else provided this terminal command result so I figured I'd share it too, lol:

bradensmacbook@Bradens-MacBook-Pro ~ % sudo kextutil /Library/Extensions/360Controller.kext

Password:
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

(kernel) Notice - new kext com.apple.driver.usb.AppleUSBHostPlatformProperties, v1.2 matches prelinked kext but can't determine if executables are the same (no UUIDs).

[image]
[image]

Unboundclassic, 2020-02-15:
I just tried 360Controller 1.0.0-alpha.2 as well and the result was the same as 360Controller 0.16.11 Notarized. Endless flashing guide button and "No devices found."

Unboundclassic, 2020-02-15:
Also not sure if it means anything... but sometimes in the preference pane the controller displayed shows up as black and sometimes it shows up as white. 

[image]

FranticRain, 2020-02-16:
For all intents and purposes, it looks like the driver is working as expected. I’ve seen dongles mess with the driver before. I don’t know why or how they do it, but they somehow prevent the driver from ever seeing the controller. If you have an opportunity to try the driver with a different dongle, I’d certainly like to know the results. 

As for the color differences in the preference pane, that’s a feature of newer versions of the driver that style the controller and the preference pane based on the theme you are using in macOS. Unfortunately last I checked, there’s a bug in detecting the theme in a preference pane application that Apple hasn’t fixed yet, so it doesn’t work quite as expected.

UnavengedAvo, 2020-02-29:
I've had similar issues with a wired Xbox 360 controller on 360Controller 1.0.0-alpha.5.

I tested with both the Hyper Drive Duo USB-C Hub, as well as the Adam CASA Hub A01.

https://www.hypershop.com/products/hyperdrive-duo-hub-for-usb-c-macbook-pro-13-and-15-2016-2017-2018?variant=41421336460

https://www.adamelements.com/www/casa_hub_a01.html

I have also tested this version of the driver:

https://github.com/360Controller/360Controller/files/3993107/360ControllerInstall.dmg.zip

- With the Adam CASA Hub A01, the guide button flashes endlessly.
- With the Hyper Drive due Hub, the guide button goes on when I plug in the controller and then dies within a couple of seconds.

When I click Enable Driver, then it fails to load with Error Code -63946981

pggndr, 2020-03-15:
I have the same problem and have tried them all...
so do I have to buy a hub?
is there no other solution?

nemesis6014, 2020-03-23:
I just updated my Mac to Catalina and was having the same issue. Got my controller to show up using Alpha 5. Here's what I did:
1. Install the Alpha 5 driver, restart, and don't open when system restarts.
2. Pair your Xbox controller through the Mac bluetooth preferences WITHOUT the USB cable plugged in. The controller shows up and pairs immediately.
2. Once your controller pairs through bluetooth, plug the USB cable into the controller and your Mac.
3. Go over to the driver in settings and you should see that the controller is now recognized
 
I turned off the bluetooth off and the controller is still working and being recognized. Unfortunately, you have to repair the device through Bluetooth every time you want to connect the controller. I am not a developer or programmer, so, please excuse the simplicity of this post.

silentmouth, 2020-03-29:
Same issue. Microsoft Xbox 360 controller connected via USB. Mac OS 10.15.3 (Catalina). MacBook Pro Mid-2012. Just says "No devices found." The controller flashes once when plugging in, that is the only response.

nemesis6014, 2020-03-29:
> Same issue. Microsoft Xbox 360 controller connected via USB. Mac OS 10.15.3 (Catalina). MacBook Pro Mid-2012. Just says "No devices found." The controller flashes once when plugging in, that is the only response.
> 
> [image]

Connect your controller through Bluetooth. You can do it easily through the Bluetooth settings. Just set your controller to pairing mode. After it connects, plug it through USB with the Bluetooth still on. Go to the xbox360 controller driver and it should show up.

juanca451, 2020-05-31:
> I just updated my Mac to Catalina and was having the same issue. Got my controller to show up using Alpha 5. Here's what I did:
> 
> 1. Install the Alpha 5 driver, restart, and don't open when system restarts.
> 2. Pair your Xbox controller through the Mac bluetooth preferences WITHOUT the USB cable plugged in. The controller shows up and pairs immediately.
> 3. Once your controller pairs through bluetooth, plug the USB cable into the controller and your Mac.
> 4. Go over to the driver in settings and you should see that the controller is now recognized
> 
> I turned the bluetooth off once I saw it show up in the driver and the controller is working through USB. I am not a developer or programmer, so, please excuse the simplicity of this post.
> 
> [image]

Hi, nemesis6014

It worked for me! But, I have no response of the buttons. It recognizes my controller, but on the Controller Test, the buttons are not responding. Any ideas?

Thanks!

Unboundclassic, 2020-08-07:
Alpha 6 fixed the issue for me! Working great now!

closed 2020-08-07 by Unboundclassic

zach777777, 2022-10-09:
I'm on Catalina. I had the same issue with my wired controller and tried Alpha 6 and many others.

The solution for me was Alpha 5. I kept my controller plugged in when I restarted my computer and now it recognizes it.

I have to say this was really annoying and I'm never updating my Mac again lol