360Controller/360Controller #1017: Thrustmaster GPX Right Stick Issue (Not game related)

Necrospasm, 2020-03-05:
### Type of Controller
Xbox 360

### OS Version
High Sierra 10.13.6

### Driver Version
Version 1.0.0-alpha.5

### Connection Method
Wired

### Device Name and Info
GPX Gamepad:
Product ID:	0x5b02
Vendor ID:	0x24c6
Version:	2.02
Speed:	Up to 12 Mb/sec
Manufacturer:	Thrustmaster

### Response of Controller
Everything works fine except the right stick. This drifts constantly all the way to one side and stays there until I press the stick down (which doesn't always work)

### Response in Preference Pane
Same as above..right stick has the same behaviour.

### Detailed Explaination of Issue
When checking with the driver preferences pane drifts to the side. This is not game specific. I have attached a gif (Hopefully it shows on github to maybe explain better). I have tested this with a windows machine as well a 360 console and can confirm that it works perfectly. Any help would be appreciated. Thank you.

[image]

FranticRain, 2020-03-05:
Can you try setting a small deadzone around the right stick and seeing if that resolves the issue? Windows and 360 both impose a default deadzone, which we do not. Adding the deadzone back into the equation will likely solve the problem. Especially seeing as the driver never interpolates movement, so all updates to stick position would be based on packets sent from the controller.

Necrospasm, 2020-03-05:
Hot damn that seems to have done it. Thanks you so much, really appreciate it! :)

Necrospasm, 2020-03-05:
Spoke too soon...it's still doing it although not as bad as before. Tried playing around with the slider. Should it be linked or unlinked?

FranticRain, 2020-03-05:
I don't believe that "linked" would make much of a difference either way with this, so you might have to make your own call on which one you prefer.

Why you say that it's not as bad, do you just mean that it happens less often? Or is there a difference in the symptoms?

FranticRain, 2020-03-06:
I think you may just have to make the deadzone a little bit bigger. The default deadzone on Window is actually about 24% of the maximum range, so I would say that you can afford to size it up, especially since it appears to be hanging right at the edge of the deadzone. If you feel like you are giving up too much of the low-end of the stick to deadzone, you might consider using the "normalize" option. Does it always hang at the edge of the deadzone no matter how large you make it?

Like I said before, I'm tempted to say that this is a controller issue that is just more obvious with 360Controller than it is a bug, as I imagine that if it was a deadzone bug, it would have been reported before. I'm not going to rule it out, but I want to make sure that we eliminate the possibility of your controller being the problem first.

Necrospasm, 2020-03-06:
yes it seems like it's hanging around deadzone...i increased the size and this is the result... normalize as well...attached are the results.
[image]
[image]

Just some more info, the controller is actually brand new, unboxed about a week ago and as I said, on windows and 360 console. Just to make sure, i tried it on a windows laptop as well and it does work well, so I'm not sure if it's actually the controller. I mean, wouldn't it be messed up in other systems too?

Necrospasm, 2020-03-06:
This is the result without the linked checked and a bigger deadzone.
[image]

FranticRain, 2020-03-07:
It's hard for me to tell what is going on in these gifs since I don't know what your real inputs are. Is this implying that it still hangs on the edge of the deadzone, even when the deadzone is huge?

Necrospasm, 2020-03-07:
You're right it's a bit difficult to deduct. So, basically when the preferences window is open, it's at the center. Then you move the right stick and it starts spazzing to the same direction of the stick, then it just hangs on the edge of the dead zone, regardless of its size.

shapeshifter87, 2020-08-09:
@FranticRain how di i link an xbox?windows controller to my macbook
struggling here