360Controller/360Controller #1018: Official wired Xbox 360 controller not found

equuscaballuss, 2020-03-08:
### Type of Controller
Replace this text with: Original Xbox 360

### OS Version
Mojave 10.14.6

### Driver Version
360Controller 0.16.11, 360Controller 0.16.11 Notarized, 360Controller 1.0.0-alpha.1, 360Controller 1.0.0-alpha.2, 360Controller 1.0.0-alpha.3, 360Controller 1.0.0-alpha.4, 360Controller 1.0.0-alpha.5

### Connection Method
Wired

### Device Name and Info
N/A

### Response of Controller
The Xbox button blinks

### Response in Preference Pane
The controller doesn't appear in the preference pane

### Detailed Explaination of Issue
The device isn't found in the preference pane and I can't find it in the system informations either. I've tried everything I could find in the forum and now I'm hopeless...

FranticRain, 2020-03-08:
Is this an Xbox 360 controller that is only wired? Or is it a wireless controller that you have connected a "Play and Charge kit" to?

equuscaballuss, 2020-03-08:
It's an Xbox 360 controller that's only wired. I didn't really knew what to say more in the detailed explanation so ask me anything and I will do my best to bring more informations. I wanted to put screenshots of my system informations but my Mac is in French. When I connect my mouse to my Mac it show "Mad Cats R.A.T. 5 Mouse" but when I put the controller in, it shows nothing... But as long as it blinks it means it's connected right ?

FranticRain, 2020-03-08:
Yes and no. That just means that it is getting power. Before the driver can see the controller and try to pair with it, the operating system needs to see it. What's happening here is that the OS can't even see the controller, so the driver is never getting a chance to see it. Is the controller connected through a dongle or hub? Or is it connected directly to your mac?

equuscaballuss, 2020-03-08:
It's connected directly to my Mac

FranticRain, 2020-03-08:
Can you confirm that it works on a windows machine or Xbox console? It might be that the cable isn't working. At this point, it isn't an issue with the driver, as not even the operating system can see it.

modosc, 2020-03-18:
i'm seeing the exact same behavior with the xbox 360 wired controller. i've got an a 2018 macbook pro running 10.15.3 with 0.16.11 of the driver. this setup worked fine at some point in the last 6-12 months but definitely isn't now. happy to help debug this however i can.