360Controller/360Controller #1020: PDP Wired Controller for Xbox One - Stealth Series | Ghost White : not found

marms24, 2020-03-15:
### Type of Controller
PDP Wired Controller for Xbox One - Stealth Series | Ghost White :
### OS Version
macOS Mojave
version 10.14.1
### Driver Version
360Controller 1.0.0-alpha.5
360Controller 0.16.11
360Controller 0.16.10
### Connection Method
Wired

### Device Name and Info
  Identifiant du produit :	0x02a5
  Identifiant du fournisseur :	0x0e6f
  Version :	1.0f
  Numéro de série :	0000B88C1659DFC6
  Vitesse :	Jusqu’à 12 Mb/s
  Fabricant :	Performance Designed Products
  Identifiant de l’emplacement :	0x14100000 / 2
  Courant disponible (mA) :	500
  Exploitation supplémentaire actuelle (mA) :	0
>Device Name:
<br/>Product ID:
<br/>Vendor ID:

### Response of Controller
nothing happen

### Response in Preference Pane
nothing happen

### Detailed Explaination of Issue
The controller is detected by my macbook and Imac on The system information panel but the controlller isnt detected by your app..

FranticRain, 2020-03-29:
Can you run this command and your terminal and tell me what the output is? It will prompt you for your password, but this is just because it is a tool built in to macOS that is trying to manually load the driver.
```bash
sudo kextutil /Library/Extensions/360Controller.kext
```

betik, 2020-03-30:
Untrusted kexts are not allowed
Kext with invalid signature (-67062) denied:

mthwtkr, 2020-04-27:
I'm getting this same issue with the same device.

kextutil output:
`Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings:
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done):
        ChatPadKeyboardEvents

(kernel) Notice - new kext com.apple.driver.usb.AppleUSBHostPlatformProperties, v1.2 matches prelinked kext but can't determine if executables are the same (no UUIDs).`

mthwtkr, 2020-04-30:
Figured I'd leave the comment here from my searches. Looks like this controller is outright not supported on the latest Mac OS kernel

closed 2020-05-15 by FranticRain