360Controller/360Controller #1023: hyperkin duke controller not recognized

bonezee, 2020-03-22:
### Type of Controller
Replace this text with:  Xbox One

### OS Version
High Sierra 10.13.6

### Driver Version
1.0.0-alpha.5

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID: 0x0652
<br/>Vendor ID:  0x2e24

### Response of Controller
Controller vibrates when plugged in, no other response. 

### Response in Preference Pane
"no devices found" is displayed on the drop down menu in the settings when the controller is plugged in. 

### Detailed Explaination of Issue
Controller not recognised in settings. I tried the build that was posted in issue #949 , but issue persists.

jm2, 2020-08-13:
FWIW my controller isn't recognized either, even running alpha.6

Based on #949, it looks like our controller has a different product ID:

`Product ID: 0x0652`
vs.
`Product ID: 0x1618`

Either there was a new revision, or it's possible that the different colored controllers have different product IDs (mine is the green SE.)

ShinyScorpio91, 2023-02-04:
I've downloaded and ran it through 7zip & WinRAR but it won't open past 360controllerInstall file. Have I done something wrong or is there a specific way to install this driver?