360Controller/360Controller #1026: Please add support for Hori Racing Wheel Switch

ak6i, 2020-03-24:
Hori Racing Wheel Switch, https://hori.co.uk/mario-kart-racing-wheel-pro-deluxe-for-nintendo-switch/ - claims PC support via Xinput

### Type of Controller
N/A (supports Xinput)

### OS Version
macOS 10.14.6

### Driver Version
0.16.11 (but I checked the source and the hardware ID isn't present in modern-rewrite)

### Connection Method
Wired
### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
Device Name: HORI Racing Wheel Switch
<br/>Product ID: 0x013e
<br/>Vendor ID: 0x0f0d

### Response of Controller
No response.  On a PC, when identified, holding the Assign button will cause the appropriate button to light up.

### Response in Preference Pane
Controller does not appear at all.

### Detailed Explaination of Issue
360Controller does not identify the device, as the hardware ID is not included in the plist.  On Windows it is detected as "Xbox 360 Controller for Windows".

ak6i, 2020-03-25:
Attached is a patch for Info.plist.  

Sorry for not submitting via a pull request, I'm a Github amateur.  It's not tested since I can't figure out how to get Xcode 6.4 working on MacOS Mojave, nor could I figure out the changes required to build 360Controller with a later version of Xcode.  I did look into it, but couldn't find the details easily as Mac development isn't my specialisation.  I hope the patch helps speed up adding support.  

[patch.diff.zip](https://github.com/360Controller/360Controller/files/4379461/patch.diff.zip)

FranticRain, 2020-03-29:
This isn't an X-Input driver, it's an Xbox controller driver. This device won't work with this driver. The device must be able to be initialized with the Xbox controller protocol. Devices that don't work with Xbox hardware are outside the scope of support of this driver.

closed 2020-03-29 by FranticRain

ak6i, 2020-03-29:
When I connect this controller to a Windows PC, it is detected as "Xbox 360 Controller for Windows" - so I believe that it is using the Xbox controller protocol.

It would be great if you could create a build with this controller included so that I can at least see if it works.  Could you please do that for me?

FranticRain, 2020-03-29:
Here you go.

[360ControllerInstall.dmg.zip](https://github.com/360Controller/360Controller/files/4399130/360ControllerInstall.dmg.zip)

ak6i, 2020-03-29:
Thanks!  Using that build the controller works in Steam, but the System Preferences page for 360Controller crashes on load.

stefo-beam2drive, 2023-07-22:
Hi, how is it looking for the M1 macs? It doesn’t wanna install on my Macbook pro m1