360Controller/360Controller #1029: Don't work with macOS 10.15.4

Immortalise, 2020-03-29:
### Type of Controller
Xbox One S

### OS Version
macOS 10.15.4

### Driver Version
Version 1.0.0-alpha.5

### Connection Method
Wired

### Device Name and Info
N/A

### Response of Controller
When connected with USB it vibrate, then no reaction.

### Response in Preference Pane
no

### Detailed Explaination of Issue
Why does mac not recognized the controller?
The version of mac is macbook pro 2017 with thunder bolt3, so I try to use a "type-c to usb converter", this works for my wired mouse.

panosl, 2020-03-29:
I uninstalled, reinstalled and it works mate. Wired as well. Xbox One (not S).

FranticRain, 2020-03-29:
You must update your original post with your controller information. A guide on how to do so can be found here: https://github.com/360Controller/360Controller#how-do-i-find-my-vendor-id-and-product-id.

Controller information is only not required for issue that do not involve controllers. For example, a feature request that isn't specific to any controller, or maybe a typo somewhere in the code or documentation. For all issues individuals have with getting a controller working, controller information is ABSOLUTELY VITAL to solving your problem. Not only does it let us know if your controller is already supported, it also lets us know that the operating system can see the controller and is a crucial step in resolving your issue.

Please notify my when you have updated your original post with your controller information, or let me know if you are unable to find it.