360Controller/360Controller #1030: No devices found - no Security & Privacy prompt

jackmrhughes, 2020-03-29:
### Type of Controller
Xbox One (bluetooth) controller x 2
Xbox One (non-bluetooth) controller x 1

### OS Version
macOS High Sierra 10.13.6

### Driver Version
360Controller 1.0.0-alpha.5

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID:
<br/>Vendor ID:

### Response of Controller
Hello! When connecting with cable only the newer controllers vibrate, the older (2015) controller doesn't vibrate. 

### Response in Preference Pane
The preference pane says 'No devices found' for all three controllers

### Detailed Explanation of Issue
I understand the way to revolve this issue is to go to Security & Privacy and allow "Drew Mills" but that isn't appearing for me. I've tried uninstalling, restarting, reinstalling, restarting multiple times and it still doesn't appear. It did appear once and I clicked allow but then nothing happened and since then it hasn't appeared. 
[image]

FranticRain, 2020-03-29:
The driver can't see the controller if the OS can't see the controller. Try a different cable or different path to the machine (if you're using a hub) to see if you can get your controller to show up in the system info.

betik, 2020-03-30:
i am also having this problem?

FranticRain, 2020-03-30:
@betik Is your problem identical? Are the controllers not found in the system info? I can't help you with this little information.

jackmrhughes, 2020-03-30:
I tried the cable my flatmate uses to charge his PS controller and it immediately worked and recognised the controller, thanks for the suggestion! I guess the cable I was using wasn't good enough. The only issue I have now is the mapping isn't recognised in game (Left 4 Dead 2) but I realise that isn't your problem. Having more success with the Steam Big Picture Mode / Tattiebogle driver, but still needs a lot of tweaking in the controller configuration.

FranticRain, 2020-03-30:
Yeah. Not all cables are created equal. Some only have the wires in them to charge things, You need the ones that send data, otherwise the device can't talk to the operating system.

Glad that you got it working. I'm going to go ahead and close this down.

closed 2020-03-30 by FranticRain

ElDuderino777, 2020-12-01:
i'm having the same issue with my 3rd party 360 controller not showing up in the systems report...it's wired, so there's no way just to use another cord