360Controller/360Controller #1031: Macbook stuck on Booting

abcde57, 2020-03-30:
Dear Andrew Mills,

I downloaded and installed your 360 controller and it ended up with my macbook stuck on booting screen. I can't go to safe mode. tried to reinstall the OS but also failed so now I ended up with Panic kernel black thing on m left side screen and the macbook keep restarting it self.

I dont have any back up for my macbook.

Could you please help me? I'm begging you pleaseee.

Please help.

### Type of Controller
Replace this text with: Original Xbox, Xbox 360, Xbox One, All or N/A

### OS Version
10.11.6

### Driver Version
360controller 1.0.0-alpha.5

### Connection Method
Replace this text with: Wired, Wireless Adapter, Bluetooth, N/A or more specific method if applicable

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID:
<br/>Vendor ID:

### Response of Controller
Replace this text with information on how the physical controller reacts. Does the guide button light up? Does it vibrate? Etc.

### Response in Preference Pane
Replace this text with information about the controller's reaction in the preference pane. Does the controller appear in the preference pane? Do all of the buttons, joysticks, and triggers act as expected?

### Detailed Explaination of Issue
Replace this text with a detailed explaination of your issue. Simply saying "it doesn't work" is a quick way to ensure that your issue is never fixed. Without detailed information on your specific issue, we cannot fix your issue. Including as much information as possible in this section increases the likelyhood of getting your issue quickly and effectively resolved.

FranticRain, 2020-03-30:
For the record, my name isn't Andrew.

Check out this guide on how to properly uninstall the driver from a machine that won't boot: https://github.com/360Controller/360Controller/issues/948#issuecomment-556142636

abcde57, 2020-03-30:
Hi! 

So sorrry for calling you with Andrew.
Thanks for the link that you sent me. 
I will try to follow those guide, but I have a question about it.
Should I type everything after the number or I type the command after the "?

Thanks

abcde57, 2020-03-30:
Hi!

I followed the step, but it keep saying no such files or directory.

Do you have any suggestions?

Thanks

FranticRain, 2020-03-30:
Looks like there's a step missing in that guide. When you boot into the recovery partition, before you open the terminal, you need to open "Disk Utility," click on your drive on the left hand side, and then select the "Mount" option from the buttons in the middle along the top, Once that is done, you can get to your disk.

If you are having issues, I'm going to need you to be very specific, but I have a pretty good guess. This isn't the kind of guide where you copy and paste commands. Every system is different and I can't provide commands that you can just copy. You're going to need to put some effort in to fixing this, because I refuse to baby you through it.

The paths start after the quotes. So for the first item on that list on my system, that is:
```
/Volumes/Internal\ SSD/Library/StartupItems/360ControlDaemon
```

This will require that you know how to use the terminal, which you can familiarize yourself with here: https://www.macworld.com/article/2042378/master-the-command-line-navigating-files-and-folders.html

When you finally find the items in the list, you can delete them with `rm -rf`. So in my case, I would have run the commands in this order:

```
cd /Volumes
ls
cd Internal\ SSD
ls
cd Library
cd StartupItems
ls
rm -rf 360ControlDaemon
```
That would have taken care of the first item. From there, to do the second I would go up two directories to get back to where I started, and then follow the next path.
```
cd ..
cd ..
ls
cd Library
cd Application\ Support
cd MICE
ls
```
and so on.

jharle, 2020-04-07:
> 
> 
> For the record, my name isn't Andrew.
> 
> Check out this guide on how to properly uninstall the driver from a machine that won't boot: [#948 (comment)](https://github.com/360Controller/360Controller/issues/948#issuecomment-556142636)

Thank you for this; I also had to boot into recovery mode after installing over an older version of the package. After the cleanup and reboot, I was able to install the new one, and have it work successfully.