360Controller/360Controller #1032: "No devices found"

RookieWookiee, 2020-03-30:
Xbox One Elite

10.11.16 El Capitan

360Controller 1.0.0-alpha.5

Wired

Controller:
  Product ID:	0x02e3
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	2.03
  Serial Number:	7EED8C6FB6E7
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Microsoft
  Location ID:	0x14200000 / 7
  Current Available (mA):	1000
  Extra Operating Current (mA):	0

Response of Controller
Controller vibrates when plugged in, but guide button does not light up.

Response in Preference Pane
No devices found

Detailed Explanation of Issue
The controller is listed in the System Prefs > USB devices, but doesn't power on and is not recognized in the control panel.

FranticRain, 2020-03-30:
Can you run this command and your terminal and tell me what the output is? It will prompt you for your password, but this is just because it is a tool built in to macOS that is trying to manually load the driver.
```bash
sudo kextutil /Library/Extensions/360Controller.kext
```

RookieWookiee, 2020-03-30:
`Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

(kernel) kxld[com.mice.driver.Xbox360Controller]: The super class vtable '__ZTV11IOHIDDevice' for vtable '__ZTV20ChatPadKeyboardClass' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The super class vtable '__ZTV11IOHIDDevice' for vtable '__ZTV22Xbox360ControllerClass' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The super class vtable '__ZTV11IOHIDDevice' for vtable '__ZTV20ChatPadKeyboardClass' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The super class vtable '__ZTV11IOHIDDevice' for vtable '__ZTV22Xbox360ControllerClass' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV22Xbox360Pretend360Class was not patched because its parent, the __ZTV22Xbox360ControllerClass, was not patchable.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV22XboxOneControllerClass was not patched because its parent, the __ZTV22Xbox360ControllerClass, was not patchable.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV22XboxOnePretend360Class was not patched because its parent, the __ZTV22XboxOneControllerClass, was not patchable.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV27XboxOriginalControllerClass was not patched because its parent, the __ZTV22Xbox360ControllerClass, was not patchable.
(kernel) Can't load kext com.mice.driver.Xbox360Controller - link failed.
(kernel) Failed to load executable for kext com.mice.driver.Xbox360Controller.
(kernel) Kext com.mice.driver.Xbox360Controller failed to load (0xdc008016).
(kernel) Failed to load kext com.mice.driver.Xbox360Controller (error 0xdc008016).
Failed to load /Library/Extensions/360Controller.kext - (libkern/kext) link error.
Check library declarations for your kext with kextlibs(8).
MacBook-Pro:~ myusername$ 
`

FranticRain, 2020-04-01:
Give this build a shot and let me know if that solves your problem: https://github.com/360Controller/360Controller/files/3812654/360ControllerInstall.dmg.zip

RookieWookiee, 2020-04-01:
So far so good! It powers on when plugged in and also is listed by default in the control panel. I hit each button and it shows each button lighting up in the graphic! I'll post again when I actually try using it to game. Thank you!

[image]

[image]

FranticRain, 2020-04-01:
Perfect. Closing this down.

closed 2020-04-01 by FranticRain