360Controller/360Controller #1033: Controller not detected

4nZwar3, 2020-03-30:
### Type of Controller
3rd Party Controller (supposedly compatible)

### OS Version
Mac OS High Sierra 10.13.6

### Driver Version
Version 1.0.0-alpha.5

### Connection Method
Wired

### Device Name and Info
>Device Name: PC/PS3/Android Gamepad
<br/>Product ID: 0x0575
<br/>Vendor ID: 0x2563

### Response of Controller
Controller turns on but does not connect properly, appears in system info app though.

### Response in Preference Panel
None

### Detailed Explaination of Issue
The controller has many ways to be turned on by means of commands since each one is specified for different devices (ex: HOME + X = Bluetooth for PC and Android), in this case the command is HOME + Y (Mac USB Wired Mode) and it is supposed to turn on the controller as a type of Xbox 360 controller because in the instruction manual indicates that this driver must be installed for use. Anyway, when you connect the controller and turn it on the right way (with the driver already installed, of course) the light stays blinking, as if still searching for the device.
Manual instructions picture:
[image]

4nZwar3, 2020-03-30:
**Note:**
I actually couldn’t find the Product ID and Vendor ID from the Mac USB wired mode, but using the Android+PC Wired mode did, and gave me different info from the same boot. If you hold the HOME button while connected it changes the led from the 2nd one to the 3rd one, changing its info in the process. Also I could found the Bluetooth info.

**Android+PC Wired mode (v2):**
	Product ID: 0x028e
	Vendor ID: 0x045e

**Bluetooth mode:**
	Product ID: 0x0402
	Vendor ID: 0x1949

FranticRain, 2020-03-31:
Give this build a shot and let me know if that solves your problem: https://github.com/360Controller/360Controller/files/3812654/360ControllerInstall.dmg.zip

4nZwar3, 2020-04-02:
It works! Thank You

closed 2020-04-02 by 4nZwar3