360Controller/360Controller #1034: Wired XBOX 360 controller: No Devices Found on Mac OS Catalina 10.15.4

alexzucca90, 2020-03-31:
### Type of Controller
Xbox 360

### OS Version
mac OS Catalina 10.15.4

### Driver Version
Version 1.0.0-alpha.5

### Connection Method
Wired

### Device Name and Info 
>Device Name:
<br/>Product ID: 0x028e
<br/>Vendor ID: 0x045e

### Response of Controller
Home button blinks once connected, then shuts off, 

### Response in Preference Pane
No Devices Found

### Detailed Explanation of Issue
Older driver (before upgrading to Catalina) worked fine. After upgrading stopped working. Removed previous driver, rebooted and installed new driver. Rebooted again. On Controller Test panel, No Devices Found.

FranticRain, 2020-04-01:
Can you run this command and your terminal and tell me what the output is? It will prompt you for your password, but this is just because it is a tool built in to macOS that is trying to manually load the driver.
```bash
sudo kextutil /Library/Extensions/360Controller.kext
```

alexzucca90, 2020-04-01:
Thanks for the quick reply: 

here is the output 

```bash
Untrusted kexts are not allowed
Kext with invalid signature (-67062) denied: /Library/StagedExtensions/Library/Extensions/3B6546EC-18CE-4DD3-BC70-A77D87CBB763.kext
Bundle (/Library/Extensions/SteamInputX.kext) failed to validate, deleting: /Library/StagedExtensions/Library/Extensions/3B6546EC-18CE-4DD3-BC70-A77D87CBB763.kext
Unable to stage kext (/Library/Extensions/SteamInputX.kext) to secure location.
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Kext rejected due to system policy: <OSKext 0x7fa60f518b50 [0x7fff853468c0]> { URL = "file:///Library/StagedExtensions/Library/Extensions/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
Kext rejected due to system policy: <OSKext 0x7fa60f518b50 [0x7fff853468c0]> { URL = "file:///Library/StagedExtensions/Library/Extensions/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
Diagnostics for /Library/Extensions/360Controller.kext:
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents
```

SCBbestof, 2020-04-01:
Same issue here, with version 0.16.11

FranticRain, 2020-04-01:
Please ensure that the driver has been allowed in System Preferences -> Security and Privacy -> General. This approval UI prompt is only present in the Security & Privacy preferences pane for 30 minutes after an alert, such as the one that appeared during installation. Until you approve the kext, future load attempts will cause the approval UI to reappear but will not trigger another user alert. You can trigger a kext load by either re-connecting the controller or by running

```sudo kextutil /Library/Extensions/360Controller.kext```

The approval UI is as follows:

[image]

SCBbestof, 2020-04-01:
It works for me now, after I have updated to Version 1.0.0-alpha.5  🎉

alexzucca90, 2020-04-01:
Works now! Thanks!

FranticRain, 2020-04-01:
Perfect!

closed 2020-04-01 by FranticRain

mrocha2020, 2020-05-11:
Hi, please help me.
When I enter this command , show this message:

Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

(kernel) Notice - new kext com.apple.driver.usb.AppleUSBHostPlatformProperties, v1.2 matches prelinked kext but can't determine if executables are the same (no UUIDs).