360Controller/360Controller #1036: Controller recognized but my mac is not registering the inputs

Hermanator09, 2020-04-02:
### Type of Controller
Xbox one standard

### OS Version
10.14.6

### Driver Version
Most recent Alpha 0.0.5

### Connection Method
wired - micro usb

### Device Name and Info

Product ID:	0x02ea
  Vendor ID:	0x045e  (Microsoft Corporation)

### Response of Controller
It has vibrated and the guide button is lit but the input test is showing no response

### Response in Preference Pane
The controller shows in the prefpane as an xbox one controller but again none of the buttons are responding

### Detailed Explaination of Issue
So I plug my controller in with the official xbox usb and attempt to use it in the preference pane but it simply doesn't register any inputs from my controller, not really sure what else to say

FranticRain, 2020-04-02:
Give this build a shot and let me know if that solves your problem: https://github.com/360Controller/360Controller/files/3812654/360ControllerInstall.dmg.zip

FiReBlUe45, 2020-04-04:
Hello,
I have the same problem and this version of 360Controller software doesnit correct the problem.
Do you have other solution?