360Controller/360Controller #1037: PowerA Spectra Pro (Xbox One) has a different Product Id than Info.plist

SolivaTechClub, 2020-04-02:
### Type of Controller
PowerA Spectra Pro (Xbox One)

### OS Version
macOS Catalina 10.15.4

### Driver Version
https://github.com/360Controller/360Controller/releases/tag/v1.0.0-alpha.5

### Connection Method
Wired

### Device Name and Info
XB1 Spectra Pro:

  Product ID:	0x592a
  Vendor ID:	0x24c6
  Version:	1.01
  Speed:	Up to 12 Mb/s
  Manufacturer:	BDA
  Location ID:	0x14130000 / 28
  Current Available (mA):	500
  Extra Operating Current (mA):	0

### Response of Controller
Nothing

### Response in Preference Pane
Nothing

### Detailed Explaination of Issue
The controller didn't worked after unboxing in my mac.
Then I've checked two old issues about PowerA controllers. I don't know if something has changed later.
Then I went to https://support.powera.com/hc/en-us/articles/360035948691-Spectra-Controller-2019-Release-Firmware-Update-v1-18-for-Mac where I found a new Firmware for that.
Once installed the new firmware in my controller it still does not works.

But I have noticed that my product  Id is different then the config file
My product Id (0x592a = 22826)
360controller product id 21546
https://github.com/360Controller/360Controller/blob/master/360Controller/Info.plist#L3434

Maybe changing the product id in the configuration could make the trick, or theres is still a issue with macOS kernel?

I don't have a Xcode environment up and running, otherwise I would test myself.

Thanks.

ezequielantunez, 2020-04-02:
I’m the same guy that created the issue. 
Could be related to this? https://developer.apple.com/support/kernel-extensions/

FranticRain, 2020-04-02:
PDP likes to have multiple PID/VID combinations for fun. I'll add it to the plist in a bit.

ezequielantunez, 2020-04-02:
The issue for PowerA controllers was fixed? Are they working now?

FranticRain, 2020-04-02:
No. But I add them anyway, just in case.

beerboy78, 2020-07-08:
I'm having the same issue with this controller as SolivaTechClub. I can see you plan to add the new device - will this happen in the next release or is there something I can try to do myself? I'm considering whether to swap the controller which would be a shame as it seems to be really high quality for the price.