360Controller/360Controller #1040: Signed Catalina alpha 5 with wireless xbox360 support

zentis, 2020-04-04:
### Type of Controller
Xbox 360 Wireless

### OS Version
10.15.3

### Driver Version
Alpha 5

### Connection Method
Wireless Adapter

### Device Name and Info
N/A

### Detailed Explaination of Issue
I know that due to kernel panics, the wireless xbox 360 support has been disabled, would it be possible to get a release with it enabled? I am fully aware that I may get some kernel panics and will test but it would be really helpful. I don't want to disable SIP by building the driver by myself and don't have the certificates to sign a kernel extension. 

Thanks again Franticrain for all the work you've done on this.

zentis, 2020-04-04:
Actually even though it's not in the build.sh of the Alpha 5 release source code, I see com.mice.driver.WirelessGamingReceiver and com.mice.driver.Wireless360Controller in my kextstat. So are they loaded and my pairing issue is unrelated?

sakrafa, 2020-04-07:
I'm also trying to connect a wireless controller to my mac running catalina, without any success.
I do experience the same pairing issue using 0.16.4/0.16.5 (i guess no catalina support) and the current alpha (no wireless controller support).
Any help would be greatly appreciated.

germanviscuso, 2020-04-22:
Same pairing issue here. Related issue here: https://github.com/360Controller/360Controller/issues/1054

Can anybody confirm whether this alpha 5 version supports wireless or not?

Using 360Controller 1.0.0-alpha.5
Macbook Pro Retina mid 2014
MacOS High Sierra 10.13.5
But in my case it's a wireless receiver:

Product ID: 0x0291
Vendor ID: 0x045e (Microsoft Corporation)
Version: 1.07
Speed: Up to 12 Mb/sec
Location ID: 0x14100000 / 9
Current Available (mA): 500
Extra Operating Current (mA): 0

FranticRain, 2020-05-15:
The wireless driver has been built in to alpha releases for a while now. Alpha 5 includes the wireless drivers. If it doesn't work, then it doesn't work. But the support is in that release.

closed 2020-05-15 by FranticRain